pybricks/support #1: Document PID

strichter, 2020-03-22:
While I have lots of software development experience, I am only doing robotics with EV3 for fun with my kids. And I have no idea what PID is and the quick google results did not give me the tight intro I was looking for.

(I am a chemist and physicist by training, so the Calculus explanation will do. ;-)

laurensvalk, 2020-03-22:
> so the Calculus explanation will do

In that case, [this might get you started](http://ctms.engin.umich.edu/CTMS/index.php?example=Introduction&section=ControlPID) for now. Equation (1) has the basic idea. In speed control (`run`, `run_time`), the error `e` is the speed error. In position control (`run_angle`, `run_target`), `e` is the angle error.

A simplified version is intended to be eventually included on the "More about Motors" page.

laurensvalk, 2020-03-22:
The implementation is very different though, avoiding almost all numerical differentiation and integration steps. On the other hand, this brings it closer to the theory than most numerical PID implementations :)

strichter, 2020-03-24:
BTW, the link helped me to get the idea.

laurensvalk, 2020-03-24:
Sounds good. I'm thinking I'll leave this issue open until we've documented it in the docs.

closed 2023-05-02 by dlech