pybricks/support #3: `EPERM` Error when setting `motor.control.limits()`

strichter, 2020-03-24:
No matter what I tried, when trying to set any limits I get an EPERM error.

My motor is initiated as follows: `motor = Motor(Port.A, gears=[8, 40])`

I tried the following combinations to set limits:

```
motor.control.limits(30, 60, 100)
motor.control.limits(160, 320, 100)   # Reported default
motor.control.limits(30, 60, 50)
motor.control.limits(60, 120, 50)
```

They all fail.

laurensvalk, 2020-03-24:
> You can only change the settings while the controller is stopped. For example, you can set the settings at the beginning of your program. Alternatively, first call stop() to make your Motor or DriveBase stop, and then change the settings.

But if you’ve already done this, it’s a bug. Just let us know.

Think of it like changing the cruise control parameters (so not the setpoint), while driving your car. The approach followed here is perhaps a bit annoying, but it avoids all kinds of unexpected behavior.  

Making the errors more verbose and helpful is on our list. It’s already done for some errors, like the picking the wrong port, but not yet for this one. We will probably also map this error to BUSY instead.

strichter, 2020-03-24:
So, here is what I have:

```
    def reset(self):
        self.upper.stop()
        self.upper.control.limits(actuation=30)
        self.lower.stop()
        self.lower.control.limits(actuation=30)
        self.upper.run(RESET_SPEED)
        self.lower.run(RESET_SPEED)
        # Stall detection.
        while (not self.upper.control.stalled() or
               not self.lower.control.stalled()):
            tools.wait(DEFAULT_WAIT_TIME)
        # Stop all motors
        self.upper.stop()
        self.upper.control.limits(actuation=100)
        self.lower.stop()
        self.lower.control.limits(actuation=100)
        # Reset angles, so that absolute angles can be used later.
        self.upper.reset_angle(0)
        self.lower.reset_angle(0)
```

And that throws the error. I will try the new beta now.

laurensvalk, 2020-03-24:
This should work with the new build.

strichter, 2020-03-24:
Works perfectly with the latest beta.

closed 2020-03-25 by strichter