pybricks/support #4: Smart Hub Support

dracode, 2020-03-24:
The pybricks.com website references:
Move Hub (BOOST)
City Hub (Powered Up)
Control+ Hub (TECHNIC)

I think Move Hub is part 88006, City Hub is 88009.  Is that correct?
What part is "Control+ Hub"?  I cannot find anything like that on the Lego website.

I have used MicroPython fairly extensively on ESP8266 -- REALLY excited to find this project!  Any thoughts on when you may have some sort of release for non-EV3 devices?

laurensvalk, 2020-03-24:
That's right. There's a few pictures on the main page, and we'll be adding pictures to the docs too.

 - `MoveHub` follows the name from LEGO. Presumably named that way because of the built-in motors.

- The second one they call just "Hub", which isn't very descriptive. Since it comes with LEGO City Trains sets, we decided to call it the `CityHub`. It will probably will appear in other sets too, but it has studs, not technic connections, so this seems like safe choice. Who knows, maybe it will catch on.

- Control+ is the LEGO name for the new Bluetooth-capable TECHNIC series. So we're calling it `CPlusHub` in code:
[image]

- Why not call it `TechnicHub`? Because this yellow [hub](https://education.lego.com/en-us/products/lego-technic-large-hub-for-spike-prime-/45601) seems to be marketed as the Large Technic Hub. Spoiler alert, it does not actually appear in any TECHNIC set. We will probably call it the `PrimeHub`, as "Spike" appears to be the brand name (like "MINDSTORMS") and "Prime" as the set name ("EV3").

And the whole broader platform seems to be called Powered Up. So that's why all attachable sensors and motors are grouped in the `pupdevices` module.

We could be all wrong though, and then we'll be stuck with it :smile: 

> I have used MicroPython fairly extensively on ESP8266 -- REALLY excited to find this project! Any thoughts on when you may have some sort of release for non-EV3 devices?

We're glad you found this page! We're working on setting up a closed group for early alpha testing. Only for the BOOST `MoveHub` for now. Do you have one?

Everything will eventually be open, but it isn't quite ready for most everyday users just yet --- we don't want to scare off beginning users.

dracode, 2020-03-24:
Thanks!  I found the Control+ Hub on the lego site now, item 22127.  The Spike hub is 45601.

> We're working on setting up a closed group for early alpha testing. Only for the BOOST MoveHub for now. Do you have one?

I do indeed and would be happy to help test.

dlech, 2020-07-18:
The 3 request hubs are all now in beta.

closed 2020-07-18 by dlech