pybricks/support #5: Complex Sample Code

strichter, 2020-03-25:
Here is a project based on the new `pybricks` API. I really love it! It is still work in progress:

https://github.com/strichter/ev3-dog

Of note:

- It contains a simple yet effective RPC Server/Client based on BT mailboxes.
- It provides a simple implementation of a thread object called a `Task`. It also provides a `TaskGroup` class.

Keep coming back regularly to see how the dog's abilities are going to grow. The ultimate goal, of course is to have the dog walk properly and maybe even run!?

 @laurensvalk @dlech I did not know where to post this to reach the right audience.

dlech, 2020-03-25:
This is the right place (we probably don't have a very big audience yet :smile:).

strichter, 2020-03-26:
Update: I am learning more about Micropython and what features it is lacking. For efficiency reasons, Micropython does not track locals() and thus cannot have an interactive debugger.

So I implemented a console/debugger that allows command execution and inspection of whatever state was passed in. This might be very handy for anyone running their code via the console.

PS: ev3dev does not ship or make available the `make` package. Weird.

laurensvalk, 2020-03-26:
Cool! 

Believe it or not, I have been building a robot based on Spot Mini last week too. It's using the new powered up platform though, so I haven't gotten to programming it yet. It has a somewhat different structure to yours, but the same kinematics.

As you've probably figured out, there is also the REPL, and the `pybricks-repl` as a shortcut.

If you run your code with the latest vscode extension, you can configure it to enter the REPL after completing your script as well for further interactive inspection, see https://github.com/ev3dev/vscode-ev3dev-browser/releases

> # These angles have been carefully measured to provide a max upright position
> # while keep the robot stable. An angle of 0 degrees is the folded up position.

As an alternative, I have been running until stalled, and then reset_angle to e.g. 120, so that the 0 position is the stretched orientation. I've found it easier to interpret other angles that way. Coming from biomechanics, then you can make flexion positive and extension negative.

> PS: ev3dev does not ship or make available the make package. Weird.

If you like, you could use docker to compile stuff faster, see ev3dev-stretch directory at https://github.com/ev3dev/docker-library

strichter, 2020-03-26:
I did not know about REPL, but that will be useful when my son debugs using Windows and VS code. I use a much lower tech stack with Linux, Emacs, NFS and two ssh sessions into the bricks. I hope I can figure out how to start REPL from within my Python program, so I can use it instead of my little console.

On the angles, that's interesting. For me, setting the angle to 0 at the known position was a simple decision. But then I do not have any preconceived notion of positive and negative angles.

On `make`, I tend to use it to orchestrate my project. So instead of writing the two shell scripts I have there, I would just create a make file with targets for my various tasks.

@laurensvalk But here is my big question: **Do you know of a physics engine, preferably with Lego support, into which I can feed the model and run some ML on it?** 

The task of teaching the dog to walk seems just tedious and it would be a great opportunity for me to learn some modern ML. I have found some research that has done this before here: https://arxiv.org/abs/1905.06077

laurensvalk, 2020-03-26:
>  I hope I can figure out how to start REPL from within my Python program, so I can use it instead of my little console.

Just add the `-i` flag:

```
brickrun -r -- pybricks-micropython -i dog/dog.py
```

This should enter the REPL after you script completes or crashes.

laurensvalk, 2020-03-26:
> Do you know of a physics engine, preferably with Lego support, into which I can feed the model and run some ML on it?

There might be several, not sure which is the best or most practical. I've tried this one a long time ago, but not with LEGO. It's free for educational and hobby use https://www.coppeliarobotics.com/

laurensvalk, 2020-03-26:
It would be kind of epic to use something like it for unit testing as well, with the hubs and motors and sensors all modeled and simulated. But that sounds like more work than writing Pybricks in the first place :smile:

strichter, 2020-03-26:
BTW, I am planning to write a `pybricks-mock` package that will at least simulate all the behavior, so you can test your code locally before having the brick do it.

I found that this becomes invaluable when the code becomes large and more abstract. I did that last year for the printer that my older son and I were developing and allowed me to "virtually print" before ever sending the code to the brick.

strichter, 2020-03-27:
@laurensvalk Today I started thinking about walking the robot and just by manually placing the legs, I know I have a balancing problem. I can find positions where the robot will stand on 3 legs, but I have not found a way to place them in a way that the robot can walk forward. Due to visual design factors we have a straight leg base size of 32 x 17 studs.

I have just installed a gyro sensor and plan to write some code that systematically tests for stable positions.

What are you doing to get the robot walking?

closed 2020-12-16 by dlech