pybricks/support #6: [Bug]ev3 Button Class not working in R3 release

johnscary-ev3, 2020-03-30:
The Button Parameters Class does not seem to work in Release R3 compared to original LEGO release of micro-python.
Now when I use "Button.LEFT_UP" to get the code for a button pressed in IR remote I get this error,

File "/home/robot/EVE3RSTORM/main.py", line 412, in 
TypeError: unsupported types for add: 'Button', 'Button'

Line of Code with error:
if butt_sum1 == Button.LEFT_UP or butt_sum1 == Button.LEFT_UP + Button.RIGHT_UP or butt_sum1 == Button.LEFT_UP + Button.RIGHT_DOWN:

After some investigation it looks like this statement to get the pressed buttons from the IR remote:

InfraredSensorPort = Port.S4
ir =InfraredSensor(InfraredSensorPort)
buttons = ir.buttons(chan)

now returns a list of literal button names instead a list of integer button codes as before.

Is this and intentional change or a bug?

laurensvalk, 2020-03-30:
> Is this and intentional change or a bug?

This is intentional.

Although all documented 1.0 features are intended to still work, some internals have changed. Here is the 1.0 documentation for that method, which still applies today in 2.0.

[image]

Internally, it was indeed a list of numbers at that point. The same applies to colors and other parameters. Now they have types so it is easier to debug your code. Also, you can do things like `print(my_scanner.color())` and it will print e.g. `Color.YELLOW` instead of an arbitrary number.

If you would rather have button codes or bit masks, you can still get the raw values from the sensor using the new [`Ev3devSensor` class](https://docs.pybricks.com/en/latest/iodevices.html#ev3dev-sensors).

johnscary-ev3, 2020-03-30:
Thanks so much for quick and clear reply on this issue.
I agree that this change to "buttons" is indeed an improvement.

As a long time test system programmer, I often tried to figure out the best way to improve a library routine without disturbing my older code which depends on it. Usually I would make a new and improved routine with similar name, and keep the old one around to support the old code.
Just a thought.  Thanks again.

laurensvalk, 2020-03-30:
Thanks for the feedback. And totally agreed - we've tried to very hard to do that almost everywhere. For example, the new `EV3Brick()` class will have all the new stuff but the old `ev3brick` stays around. 

So please do keep sharing your findings of running your existing code on the new release, we appreciate it. There is even a specific issue tag for it now.

This one seems resolved though, so I think we can close it.

closed 2020-03-30 by laurensvalk