pybricks/support #8: i2c memory read

chrisbuerginrogers, 2020-03-31:
**Describe the bug**
If I read the i2c memory - I get inconsistent results.  If I read the memory buffer from the NXT ultrasonic sensor beginning (00) or from 2 bytes in (02) I should get the same value for bytes 2..10 - right?  thanks (note reading 0x42 does indeed give me the distance, but reading 2 bytes from address 0x40 does not).

**To Reproduce**
go to the REPL and type

>>> address=0x02 >> 1
>>> device = I2CDevice(Port.S4, address)
>>> device.read(reg = 0x00, length = 10)
b'V1.0\x00\xff\xff\xff\xff\xff'
>>> device.read(reg = 0x02, length = 10)
b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'
>>> 

**Expected behavior**
the second string I think should start with a decimal point not FF

**Screenshots**
none needed

dlech, 2020-04-02:
I think this is just the way the NXT Ultrasonic sensor works. The firmware running on the sensor itself doesn't know how to read from an arbitrary register (I confirmed this using `/usr/sbin/i2cdump` on the EV3). 

I had better luck using a HiTechnic sensor and it worked in `pybricks-micropython` as expected.

laurensvalk, 2020-04-24:
This is identified as a specific issue for the NXT ultrasonic sensor, so I'll close this issue.

closed 2020-04-24 by laurensvalk