pybricks/support #9: [Question] Remote (ColorDistance)

GianCann, 2020-03-31:
https://docs.pybricks.com/en/latest/pupdevices.html#pybricks.pupdevices.ColorDistanceSensor.remote

There is not option to manage the speed of the LPF1 motors?

laurensvalk, 2020-03-31:
Not right now, but that does not mean it won't be there in the future. I think we might have a function or class similar to the `DCMotor`, but specific to remote-controlled Power Functions 1.0.

For now, you have full control over that mode in the `LUMPDevice` class. You can easily write a function for it to encode the number. It's the 8th mode (so number 7), and the value is one 16-bit value.

Writing a function in Python is a lot easier that writing bitwise operators in the Powered Up app :smile:

laurensvalk, 2020-03-31:
Basically, all features for EV3 are pretty much final and won't change much any more. But everything for Powered Up is still expected to change quite a bit during the alpha and beta stage.

GianCann, 2020-03-31:
Thank you @laurensvalk

closed 2020-03-31 by GianCann