pybricks/support #10: [Question]Autocomplete Functions on VSCode

Sunrise98, 2020-04-04:
I want to use autocomplete functions for coding of EV3Micropython, but can't use.
Is there are way to do it on VSCode?

dlech, 2020-04-04:
With the disclaimer that we haven't focused on this use case yet so there isn't an easy way to do it, here is one way to do it right now...

1. Make sure you have [Python 3](https://www.python.org/downloads/) installed on your computer.
2. Install the [Microsoft Python](https://marketplace.visualstudio.com/items?itemName=ms-python.python) and [LEGO EV3](https://marketplace.visualstudio.com/items?itemName=lego-education.ev3-micropython) extensions in VS Code.
3. Create a new project in VS Code using the LEGO EV3 extension.
4. Open a terminal in VS Code and create a virtual environment in your project. (On Windows use `py -3` instead of `python3`.)

    python3 -m venv .venv

5. Open the `main.py` file. VS Code should ask you if you want to use the virtual environment that you just created. If it doesn't, select it in the status bar (see Microsoft Python link above for screenshots).
6. Use the command palette in VS Code to open a new Python Terminal.
7. In the Python terminal run the following

    python -m pip install --upgrade pip
    pip install git+https://github.com/pybricks/pybricks-api@master

Sunrise98, 2020-04-05:
Thank you for your quick response!
I did above way and checked the autocomplete function.

I would be grateful if you could add this function as a default.

Looking forward to your support, thanks.

devinbreise, 2020-04-12:
+1 on incorporating Python and MS Python extension into the standard installation (or at least the installation instructions).  Using the approach worked fine on my windows machine, but it is of course rather fiddly for FLL.  I wonder if this can be scripted as part of the "create a new project" Activity in the Lego VSCode Extension?

RPapendiek, 2020-05-12:
Hello David,

what is your opinion on this approach: 
https://micropy-cli.readthedocs.io/en/latest/index.html#

The micropy stub for pybricks is also available:
https://github.com/BradenM/micropy-stubs/tree/master/packages/pybricks/v1.0.0

I have not yet tested it ... but looks very promising to me ...

dlech, 2020-05-13:
Look interesting. Let us know how the testing goes.

pcgarci, 2020-06-02:
> Con el descargo de responsabilidad de que aún no nos hemos centrado en este caso de uso, por lo que no hay una manera fácil de hacerlo, aquí hay una forma de hacerlo ahora ...
> 
> 1. Asegúrese de tener [Python 3](https://www.python.org/downloads/) instalado en su computadora.
> 2. Instale las extensiones [Microsoft Python](https://marketplace.visualstudio.com/items?itemName=ms-python.python) y [LEGO EV3](https://marketplace.visualstudio.com/items?itemName=lego-education.ev3-micropython) en VS Code.
> 3. Cree un nuevo proyecto en VS Code usando la extensión LEGO EV3.
> 4. Abra el `main.py`archivo para activar la extensión de Python en VS Code.
> 5. Abra una terminal en VS Code y cree un entorno virtual en su proyecto. (En Windows use en `py -3`lugar de `python3`).
>    ```
>    python3 -m venv .venv
>    ```
> 6. VS Code debería preguntarle si desea utilizar el entorno virtual que acaba de crear. Responde sí. Si no es así, seleccione el entorno virtual como intérprete de Python en la barra de estado (consulte el enlace de Microsoft Python arriba para ver capturas de pantalla).
> 7. Use la paleta de comandos en VS Code para abrir una nueva terminal de Python (busque el `Python: Create Terminal`comando).
> 8. En la terminal de Python ejecuta lo siguiente
>    ```
>    python -m pip install --upgrade pip
>    pip install git+https://github.com/pybricks/pybricks-api@master
>    ```
> 9. _EDITAR: Este paso fue necesario en el LEGO EV3 MicroPython v1.0.0 pero ya no es necesario en v2.0.0 (lanzado en mayo de 2020)_
>    En el `main.py`archivo, cambia la línea
>    ```
>    from pybricks import ev3brick as brick
>    ```
>    
>    
>    a
>    ```
>    from pybricks.hubs import EV3Brick
>    ```
> 
> La finalización del código ya debería estar funcionando cuando realice este cambio.

Does this still work? I have not been able to carry out this process

pcgarci, 2020-06-02:
Does this still work? I have not been able to carry out this process
i have this error
 Running command git clone -q https://github.com/pybricks/pybricks-api 'C:\Users\ANGELA PATIÑO\AppData\Local\Temp\pip-req-build-ibu3fuen'        
  ERROR: Error [WinError 2] El sistema no puede encontrar el archivo especificado while executing command git clone -q https://github.com/pybricks/pybricks-api 'C:\Users\ANGELA PATIÑO\AppData\Local\Temp\pip-req-build-ibu3fuen'
ERROR: Cannot find command 'git' - do you have 'git' installed and in your PATH?

dlech, 2020-06-03:
> Cannot find command 'git' - do you have 'git' installed and in your PATH?

It requires that `git` is installed on your computer and the path to `git` is in the `PATH` environment variable.

You can get `git` at https://git-scm.com if you don't have it already.

pcgarci, 2020-06-03:
I already downloaded the git, and added the path to the PATH, however the error persists, I don't know what to do.

debruler-jl, 2020-06-03:
It seems that I've done all the steps correctly per dlech above, with no errors, but still autocomplete does not work.  I've verified pybricks 2.0.0 is installed with "pip list".  Any ideas?

laurensvalk, 2020-06-07:
In your project, go to `.vscode/settings.json`.

Delete or comment out the lines about linting and the language server:

```c
{
	"files.eol": "\n",
	"debug.openDebug": "neverOpen",
	// "python.linting.enabled": false,
	// "python.languageServer": "None"
}

```

Does that help?

debruler-jl, 2020-06-08:
Commenting out those lines in the settings.json worked!  (After a restart and it complaining about not having a linter installed, but I’m ignoring those now.) Thank you!!  The kids are much happier to have the auto complete working, it seems like we are spoiling them :-)

laurensvalk, 2020-06-08:
Thanks for getting back.

Those settings are there by default because for most users, this avoids the complexity of having to set it up. (Otherwise, users would always see red squiggly lines unless they install extras.)

But it looks like we should update the step by step guide above. (Or find a way to make it easier.)

devinbreise, 2020-10-13:
I think I'm missing something simple, or perhaps I want too much...

My FLLers are doing great and cranking Pybricks code left and right, so its time to help them stay organized.  Instead of a single flat file structure, I'd like them to organize something like this:

PROJECT
---|---lib
---|---|--- file.py
---|---|--- file2.py
---|--- missions
---|---|--- mission1.py
---|---|--- mission2.py
---|--- test
---|---|--- test1.py
---|---|--- test2.py
---|---main.py

from main.py, imports and autocomplete are intact.  for example:
`
from lib.file import SomeClass
`

But when we try to import modules/symbols from lib into a parallel directory (like missions or test), issues appear.  Attempts to use absolute paths like the one above generate "unable to open" complaints from pylint and attempts to use relative paths
`
from ..lib.robot import SomeClass
`
generate "Attempted relative import beyond top-level package" complaints from pylint.

Weirdly, autocomplete then works on instances of SomeClass in that file.

However, when a class in mission1.py inherits from a class in lib.file.py, autocomplete is dead again.

StackOverflow failed me on this one...thanks in advance for any pointers here

dlech, 2020-10-13:
Does it help if you put empty `__init__.py` files in all of the directories so that Python sees them as modules?

devinbreise, 2020-10-14:
Like magic...thank you!

Also, another tip for those reading this thread.   A couple of Python "type hints" strategically applied can make a huge difference for the kids.  A team I coach created a "Robot" class that "knows" about their particular robot design and gave it lots and lots of useful methods for doing common actions.  When they passed that into the __init__ method on their "mission" class, it lost it's typing information, and with it went autocomplete on all those methods...  By adding the type hint:
`
    def __init__(self, the_robot: Robot):
`
they are now back in their happy place.

lobodpav, 2021-12-30:
@devinbreise Started coding for Mindstorms Inventor Hub very recently. I come from the Java world and the idea of having the whole code in a single file is terrifying. When organising your project in Python modules, how do you upload it to the Hub?

laurensvalk, 2021-12-30:
This topic was originally opened when there was only the EV3, which supports multi-file projects.

We don't support this yet on Pybricks for the MINDSTORMS Inventor Hub.

lobodpav, 2021-12-30:
@laurensvalk That might be a decent workaround. Thanks!

thesynman, 2021-12-31:
> @devinbreise Started coding for Mindstorms Inventor Hub very recently. I come from the Java world and the idea of having the whole code in a single file is terrifying. When organising your project in Python modules, how do you upload it to the Hub?

I had the same concerns. My solution was to extend pybricksdev with a pre-processor that can combine multiple files into one, by following the import statements. It worked pretty well and I used it a lot, but not for a few months. I did create a pull request to integrate it with the master branch (see https://github.com/pybricks/pybricksdev/pull/20) but I never managed to get it over the line to be merged. It looks as though it now needs some updates to be compatible with the latest pybricksdev. Feel free to adapt it if it would help you.

lobodpav, 2021-12-31:
@thesynman Thanks for sharing this! From what I've seen the requested changes are simple cosmetics easy to fix. Or do I miss something that is more difficult to do to get it over the line?