pybricks/support #11: Using Mindsensors NXTMMX with pybricks

voraze, 2020-04-05:
I am trying to use the Mindsensors NXTMMX (motor multiplexer) with the EV3 using micropython.  The NXTMMX is powerd and plugged into S1.

The EV3 sees ports in1:i2c3:M1 & in2:i2c3:M2 but I am having trouble declaring motors  to those ports.

Here is the beginning of my code;

#!/usr/bin/env pybricks-micropython
 
# from pybricks import ev3brick as brick
from pybricks.ev3devices import (Motor, TouchSensor, ColorSensor,
                                 InfraredSensor, UltrasonicSensor, GyroSensor)
from pybricks.parameters import (Port, Stop, Direction, Button, Color,
                                 SoundFile, ImageFile, Align)
from pybricks.tools import print, wait, StopWatch
import struct

# Declare motors 
left_motor = Motor(Port.B)
right_motor = Motor(Port.C)
lift_motor = Motor(Port.A)
grab_motor = Motor(Port.D)
reach_motor = Motor(Port.in1:i2c3:M1)

What do I need to do to be able to utilize the motors at these ports?

dlech, 2020-04-05:
The Pybricks motor classes only support the built-in motor ports on the EV3.

It is still possible to use the NXTMMX as a generic I2C device. See: https://docs.pybricks.com/en/latest/iodevices.html#i2c-device

Sunrise98, 2020-04-12:
I want to use EV3 Sensor Multiplexer on pybricks ver.2.0.
I could use it on pybricks ver 1.0 using ev3dev2.sensor.lego, but error was happened on ver.2.0.
Could you please advise how to do?

laurensvalk, 2020-05-04:
Just like the NXTMMX, the EV3 Sensor Multiplexer is an I2C device. Because of that, sensors and motors work quite differently than when they are plugged in directly.

You can still use them using the [I2CDevice class](https://docs.pybricks.com/en/latest/iodevices.html#i2c-device).

Mindsensors already provides device drivers for some languages (RobotC, NXC, LeJOS, etc...). Perhaps you can ask them to add support for Pybricks too?

They can do that by making a Python file that uses the `I2CDevice` class. Then you can just include that file in your project.

Sunrise98, 2020-05-05:
I tried to connect using Ev3devDensor and I2CDevice class, like following codes.
```
from pybricks.iodevices import Ev3devSensor, I2CDevice

mux = Ev3devSensor(Port.S1)
# mux = I2CDevice(Port.S1, 0xA0 >> 1)
```
And this Error is occured in both cases.
```
robot@ev3dev:~$ brickrun my_project/mux.py 
Traceback (most recent call last):
  File "my_project/mux.py", line 14, in <module>
OSError: [Errno 5] EIO: 

Unexpected hardware input/output error with a motor or sensor:
--> Try unplugging the sensor or motor and plug it back in again.
--> To see which sensor or motor is causing the problem,
    check the line in your script that matches
    the line number given in the 'Traceback' above.
--> Try rebooting the hub/brick if the problem persists.
```
Please tell me the collect way to use multiplexer from mindsensors.com.

dlech, 2020-05-05:
`Ev3devSensor` cannot be used with the NXTMMX or with any sensors connected to the NXTMMX. Use `I2CDevice` only.

Sunrise98, 2020-05-05:
Hi, dlech.
Sorry, this is the question for sensor multiplexer from mindsensors.com, so I should have post as new question.
I also tried to use I2CDeviceSensor class (change the comments in above codes) and same error (OSError) was occured.
For your reference, I can confirm the connection in Device Browser normally but I can't connect on pybricks.
(If you would like, I want you to add class for sensor multiplexer as same as other language.)

dlech, 2020-05-05:
Yes, it would be good to start a new issue for the sensor multiplexer. But the answer is still basically the same.

Nikolay-Zagrebin, 2020-06-11:
When using the Mindsensors NXTMMX (motor multiplexer) with the EV3 the error is the same

#!/usr/bin/env pybricks-micropython
from pybricks.hubs import EV3Brick
from pybricks.parameters import Port
from pybricks.iodevices import I2CDevice

# Create your objects here.
ev3 = EV3Brick()

# Initialize I2C Sensor
device = I2CDevice(Port.S4, 0x06 >> 1)

-----------------
Starting: brickrun --directory="/home/robot/NXTMMX_Python" "/home/robot/NXTMMX_Python/main.py"
----------
Traceback (most recent call last):
  File "/home/robot/NXTMMX_Python/main.py", line 11, in <module>
OSError: [Errno 5] EIO: 

Unexpected hardware input/output error with a motor or sensor:
--> Try unplugging the sensor or motor and plug it back in again.
--> To see which sensor or motor is causing the problem,
    check the line in your script that matches
    the line number given in the 'Traceback' above.
--> Try rebooting the hub/brick if the problem persists.

----------
Exited with error code 1.

Nikolay-Zagrebin, 2020-06-19:
Question to the developers.
In **ev3dev2** I can do this:

#!/usr/bin/env python3
from ev3dev2.motor import LargeMotor, SpeedPercent
m = LargeMotor("in1:i2c3:M2")
m.on_for_rotations(SpeedPercent(100), 10)

This way I can access the tacho-motor class methods.
Perhaps there is a similar trick in pybricks-micropython?

And tell me, I can't find a description of the pybricks-micropython API of the Port class.
I only found this

class Port(_PybricksEnum):
    """Port on the programmable brick or hub."""

    # Generic motor/sensor ports
    A = ord('A')
    B = ord('B')
    ....

As I understand this description of system ports, but when connecting Mindsensors NXTMMX, ports are also created, which means you can do the same in pybricks-micropython. At least that's what I think.

laurensvalk, 2020-06-19:
Thanks for your message. There is no built-in class for using that device, and we will probably not add one. I'm afraid we can't possibly support all third-party vendor devices.

Instead, we provide the generic tools, such as the [I2CDevice](https://docs.pybricks.com/en/latest/iodevices.html#i2c-device) class. This will make it quite easy for the people at mindsensors to support it. Basically, it is a matter for [mapping the right methods to the right I2C commands](https://github.com/botbench/robotcdriversuite/blob/master/include/mindsensors-motormux.h).

They already do that for other programming languages. So unless someone wants to do this themselves, your best bet is to ask mindsensors support. Perhaps you can give them the link to this issue so they know what to do.

We're happy to help and explain how they can use the `I2CDevice` class and share it with their customers.

Nikolay-Zagrebin, 2020-06-19:
Thank you very much

closed 2022-12-07 by laurensvalk