pybricks/support #12: [Feature] : A simple interface for changing the Brick Name for Bluetooth connections

devinbreise, 2020-04-09:
Our lab will be using multiple EV3s and multiple PCs.  To help the kids keep track of which programmer is connected to which brick, we would benefit from a kid-friendly way to rename the bricks in the Bluetooth address space.  We have found the instructions below, but we would prefer not to teach the kids sudo... ;)

Current workaround: In an SSH terminal, run `sudo ev3dev-config`, the go to `Advanced Options` and select `Hostname`. There seems to be a bug after this where the menu program disappears. Just press ENTER and it will come back. Then you can type in the new name.

dlech, 2020-04-09:
Hi, this comes from the ev3dev project and is out of scope for Pybricks (we just do MicroPython).

There is an issue for this already that you can subscribe to https://github.com/ev3dev/ev3dev/issues/1251

devinbreise, 2020-04-09:
Got it.  Thank you!

closed 2020-04-09 by devinbreise