pybricks/support #15: [Feature] Add separate accessors for drive speed and turn rate to DriveBase

devinbreise, 2020-04-14:
Writing some code to determine maximum drive_speed for a particular robot, needed to poll DriveBase.state() in a tight loop.  Didn't really need the other 3 return values.

distance() and angle() have separate accessors.  It might make for cleaner code in some situations to have the other two as well.

laurensvalk, 2020-04-15:
Thanks for your input.

This is a bit of a trade-off between convenience, and minimizing the overall number of methods. We try to to avoid duplicate functions, except where it would be used by many users. 

We felt that `angle` and `distance` would be the most common ones. Speed and turn rate less so, because those are the ones being set.

This keeps the docs easier to read, but it also reduces the amount of storage needed to implement it. The EV3 has plenty of space, but some of the other programmable hubs do not.

As you might know, you can also do this:

```python
s, v, a, w = db.state()
````

```python
_, speed, _, _ = db.state()
````

```python
speed = db.state()[1]
````

By the way, because the DriveBase is new in 2.0, I'm very curious to get your feedback.

We still have to select the right default settings (`settings()`) but also the parameters in the `control` attributes that govern the default accelerations for `drive`. So if you've seen something that doesn't feel right, please do share. Thanks!

devinbreise, 2020-04-15:
Got it.  The return() of Tuples in Python is interesting...leads to discussions like this.  If in the interest of complete but minimal API, my vote would be for individual accessors making the state() method  redundant, but I understand its a style choice.
> By the way, because the DriveBase is new in 2.0, I'm very curious to get your feedback.
Thanks for asking.  I have some, I'll post in another "thread".

laurensvalk, 2020-04-20:
>  If in the interest of complete but minimal API, my vote would be for individual accessors making the state() method redundant, but I understand its a style choice.

The benefit of `state` is that it reads all values at once, so code runs faster. More so, they are read at the same point in time, which can be helpful if you use it to build some kind of controller.

So basically we have one quick does-all method for advanced users, plus a few convenience functions for beginning/moderate users.

As there is no further action to take here, I'll go ahead and close this issue.

closed 2020-04-20 by laurensvalk