pybricks/support #17: [Bug] DriveBase.settings() not changing turn_acceleration value

devinbreise, 2020-04-16:
Was trying to find acceleration thresholds and found this:

This code (drive is an object of type DriveBase):
```Python
        print("Default:"+str(self.drive.settings()))
        self.drive.settings(10, 20, 30, 40)
        print(self.drive.settings())
        self.drive.settings(turn_acceleration=40)
        print(self.drive.settings())
 ```
Produces this output:
Default:(100, 200, 90, 180)
(10, 20, 30, 30)
(10, 20, 30, 30)

laurensvalk, 2020-04-16:
That’s a bug! Found it and will be fixed in next release. Thanks!

laurensvalk, 2020-04-20:
This should be fixed by https://github.com/pybricks/pybricks-micropython/commit/f641da15f49a3cb9776841d0c7735ed58f8ea8bb

We will make that repository public in a few weeks. For now, if you'd like to test, please do the following:
- Download the [`pybricks-micropython-ca65b2f`](http://pybricks.com/wp-content/uploads/build/pybricks-micropython-ca65b2f) which has the fix
- If you use Mac or Linux, run `chmod +x pybricks-micropython-ca65b2f` to make it executable
- Add it to your project folder.
- Tell your `main.py` to use this local version instead by changing the first line to:

```python
#!/usr/bin/env /home/robot/my_project_name/pybricks-micropython-ca65b2f
```

Your other projects will be unaffected.

devinbreise, 2020-04-20:
Thanks! happy to try the mods in this build but I'm getting a 404 on this url:
http://pybricks.com/wp-content/uploads/build/pybricks-micropython-ca65b2f

laurensvalk, 2020-04-20:
I’m not sure what went wrong there, you’re right.

[pybricks-micropython.zip](https://github.com/pybricks/support/files/4506136/pybricks-micropython.zip)

Does this work? Unzip and then the same applies as above.

devinbreise, 2020-04-20:
Sorry to use your time with this...
Download was no problem.  
I noticed the zipped file did not have the build number on the end so I added "-ca65b2f" to it and placed in the project folder next to main.py.  Removed the current #! directive in main.py and put in the one from above with my_project_name replaced appropriately.
Download and Run from VSC on Windows produced this:
```
Starting: brickrun --directory="/home/robot/coach_test" "/home/robot/coach_test/main.py"
----------
Starting program failed: Failed to execute child process "/home/robot/coach_test/main.py" (Permission denied)
----------
Exited with error code 1.
```
Here is a little more info from the unix shell on the EV3

```
robot@ev3dev:~/coach_test$ ls -l
total 600
-rw-rw-rw- 1 robot robot   1091 Apr 11 01:22 main.py
-rw-rw-rw- 1 robot robot   2983 Apr 11 01:22 myMenu.py
-rw-rw-rw- 1 robot robot   4118 Apr 11 01:22 my_arms.py
-rw-rw-rw- 1 robot robot    844 Apr 11 01:22 my_display.py
-rw-rw-rw- 1 robot robot  10673 Apr 11 01:22 my_drive.py
-rw-rw-rw- 1 robot robot    630 Apr 11 01:22 my_tools.py
-rwxrw-rw- 1 robot robot 567256 Apr 11 01:23 pybricks-micropython-ca65b2f
```

dlech, 2020-04-20:
My guess is that the shebang is not right, e.g. there is a blank line on extra characters before it.

You can also run `chmod +x main.py` on the EV3 to fix the permissions.

devinbreise, 2020-04-20:
Good guess.  left a blank line before the shebang.  Very fiddly this Python world!  Thank you and my apologies again.

devinbreise, 2020-04-21:
Testing with ca65b2f I can see that the bug mentioned above does seem to be fixed.  Thanks!  However, I am still seeing something I don't fully understand.

I'm testing turn() on RobotDrive using settings() with a set turn_velocity around 250 and adjusting the turn_acceleration parameter.  My goal was to discover, for the robot I'm testing, at what threshold of turn_acceleration the turn() maneuver becomes inconsistent (due to wheel slippage).  This is an experiment we take FLL teams through all the time, though usually with varying rates of velocity instead of acceleration because the (steep!) acceleration curves are fixed in the MindStorms blocks.  Code looks like this:
```Python
            self.drive.settings(turn_acceleration=maxAccel)
            self.drive.turn(360)
            self.drive.stop()
            self.hold() #this calls hold() on the two drive motors
```
The call to hold() is to "lock" the motors in case there is any residual momentum. 

What is surprising to me is that I can't seem to "break" it.  As I increase the turn_acceleration value, I  do see a shorter and shorter time to complete a turn(360) but it seems to cap out at a certain level, while it is still showing a high degree of accuracy.  I thought perhaps the underlying Motor.control.limits were creating this cap, so I set the acceleration value there on both drive motors to a very high value, but that didn't seem to affect this cap.

So the good news is that the software is exhibiting a remarkable combination of speed/precision (360 degree spin in ~1.5 seconds with 1-2 degree accuracy with a medium sized robot).  Very impressive.

But I feel there is still something in the software prohibiting me from getting it to the point where it loses accuracy.  Perhaps the pid settings on the two wheels?

laurensvalk, 2020-04-21:
> Testing with ca65b2f I can see that the bug mentioned above does seem to be fixed. Thanks!

Glad to hear it!

I hope you don't mind I moved your new findings to a separate issue in #20. I'd like to close this bug report as this particular one is fixed.

closed 2020-04-21 by laurensvalk

m-isn, 2025-02-25:
robot@ev3dev:~/ev3_wifi$ chmod +x main.py
robot@ev3dev:~/ev3_wifi$ ls -l
total 8
-rw-rw-rw- 1 robot robot 2117 Feb 25 14:39 example.txt
-rwxrwxrwx 1 robot robot 2201 Feb 25 14:39 main.py <-- (after change file permission)
robot@ev3dev:~/ev3_wifi$ ls -l
total 8
-rw-rw-rw- 1 robot robot 2117 Feb 25 14:41 example.txt
-rw-rw-rw- 1 robot robot 2201 Feb 25 14:41 main.py <-- (this after redownload main.py permission back to previous sets)
robot@ev3dev:~/ev3_wifi$