pybricks/support #18: [Project] Testing Line Following with default DriveBase.limits()

devinbreise, 2020-04-16:
I tested a Lego robot with a 95mm axle track and caster wheel on the back using the default limits() (currently distance_control (349, 698, 100) and heading_control ((349, 698, 100)

I was able to get the robot to reliably follow a fairly irregular line (the one that extends into base on this season's CityShaper mat) with velocity 50 and a p coefficient of 1.5 applied to a raw reflected light sensor reading.  Faster speeds are possible, but the sharpest corners won't make it.  For a straight line, it would be possible to increase the velocity somewhat.
```Python
self.drive.drive(50, (self.lightL.reflection() -65) * -1.5) #65 to account for offset light sensor relative to axle track
```
Observations:  
--the oscillations due to the proportional control are fairly significant.  The loop seems to want a integral control.  However, it is comparable to MindStorms performance.
--the loop rate is very fast (measured about 1.3ms per loop), but the drive doesn't seem to be responding that quickly.  Movements from side to side appear to be something more like 200ms.  Here is a video:  https://drive.google.com/file/d/1CX54gm24_DL3t5Zx-Q3sONnfOJakDFdR/view?usp=sharing

I'd be very interested on thoughts regarding loop rate versus drive response.

laurensvalk, 2020-04-16:
Yes, it looks like it’s being limited by the drive base acceleration, but apparently that can’t be changed right now due to #17, oops!

I suppose in theory and with some math you could flip the direction of one motor and invert the meaning of drive and turn 😄 .

I’ll make sure to get you a fixed build this weekend.

laurensvalk, 2020-04-16:
Scratch my comment above - those settings aren’t used by drive, so you can still change the acceleration through the heading control attribute.

devinbreise, 2020-04-16:
I'll admit I'm still a little lost when it comes to understanding the relationship between setting a cap on angular velocity acceleration and a PID (or just a P) controller.  My instinct here was to set the angular acceleration rate to a very high number to essentially disable it and let the Proportional control essentially handle that by tying angular velocity to the sensor reading in a proportional way.  But I see that the default is already pretty high (698 deg/sec2) so is the idea that one should lower that number to somehow shorten the oscillation period?

laurensvalk, 2020-04-16:
The idea is that the acceleration setting should not play a role, so kids don’t have to worry about it. But right now it appears that it does, and possibly other settings too. I’ll work on a few initial defaults and share them to try out.

The current values are quite arbitrary so I wasn’t expecting them to work great for line followers.

A P controller on the light value should be sufficient. The drive base has its own controls to ensure that this speed is tracked. But clearly not smoothly enough right now.

laurensvalk, 2020-04-20:
Preliminary testing shows that the limits might be fine. It does not really seem to hit them, but I’ll look in more depth tomorrow.

It looks like it is currently not very quick to track the drive speed and turn rate because the control gain defaults are a bit low. Try doubling the kp gain in heading control pid (From 200 to 400) for now. This makes it track your turn rate more quickly, which makes it more responsive.

I’ll try to build a few drive bases and include reasonable defaults in the next release.

devinbreise, 2020-04-22:
I spent a little more time trying to tune a fast and highly responsive line follow.  I continue to see control loops with ~3ms loop time (polling a light sensor, a bit of math, and adjusting two motors) which is really encouraging...haven't found the right coefficients yet...

However, I'll admit I'm still perplexed by the following:  Do you have some typical scenarios in mind for how you would expect someone to code a line follow using Pybricks?

At the moment, because I don't really know the answer to that question, I'm bypassing (I think) all of the PIDF code in Pybricks and using Motor.duty() inside my own PID controller loop.  But I suspect you have something else in mind (based on these comments) that I don't really understand.  

> A P controller on the light value should be sufficient. The drive base has its own controls to ensure that this speed is tracked. 

Perhaps there is some assumption about how if someone using Pybricks did a simple proportional translation from a sensor reading to a commanded robot rotational velocity, the Pybricks controller(s) would come into effect in a way that would create the desired outcome? If so, would you be so kind as to suggest what that connection between the light sensor and the Pybricks drive() API would be?  Would you tie it to turn_rate and leave drive_speed constant?  Or maybe tie it to both in an inverse way?  Or did I just make this whole theory up?  ;)

laurensvalk, 2020-04-22:
I think what you said earlier was just fine:

> My instinct here was to [...] let the Proportional control essentially handle that by tying angular velocity to the sensor reading in a proportional way.

And you can make it even better by also slowing the drive speed proportionally. Then it slows down in difficult corners.

laurensvalk, 2020-04-22:
I think we'll include a sample in the docs. Both the basic one, and maybe also the one which slows down in difficult corners.

laurensvalk, 2020-04-22:
I tried this earlier today with the default educator bot on a fairly challenging course, and this was going well.

It is only when you want to go much faster, and you still want to use the `DriveBase`, you may have to increase `kp` as mentioned to make sure it quickly responds to your changing commands.

For full control, indeed, there's always `dc`. That's just fine, but it's much more work (and harder for kids) to then extended it with things like slowing in corners.

devinbreise, 2020-04-22:
Thank you!  I'll give this a try in several ways.  I like this from an educational point of view because it provides for a nice cognitive glide slope into the mysteries of PID without having to jump from a simple P controller to writing your own PID code...which, as you say, is a big leap.

laurensvalk, 2020-04-22:
Maybe side tracking here, but

> provides for a nice cognitive glide slope into the mysteries of PID

I recommend the `track_target()` in the `Motor` class for that lesson.

Let that motor carry a small weight on a lever. Then play with the PID constants.

It's much easier to explain and analyse PID on something with a stationary target than a moving one.

devinbreise, 2020-04-23:
Well, try as I might, I haven't been able to coax the PyBricks controller to dampen the oscillation on a line follow.  I'll be interested in seeing what settings you come up with when you have a chance.  FWIW, I was fairly successful with write your own PID on top of duty() approach.

laurensvalk, 2020-04-24:
Here's a basic example:
https://github.com/pybricks/pybricks-projects/blob/master/official_models/ev3/education_core/robot_educator_line/main.py

It's using the default LEGO Education robot line follower. I configured it for this basic track:

[image]

devinbreise, 2020-04-24:
Thanks!  No surprises in the code or set up.  However, I am curious, when you run this, do you see the controller dampening the oscillation on the straight portions?

laurensvalk, 2020-04-24:
Yes, it's smooth enough, but perhaps I didn't pick a very high drive speed (100) or maybe this line is less complicated.

I couldn't be sure from your video, but it looks like the light sensor is quite close to the wheels and it has all the inertia behind the wheels. Although... That's not so different from the standard educator bot.

I'm not sure how you are connected to your robot, but if you configured it for internet access, you can do this to update it to the latest version from two days ago just to make sure:
- Go to the ev3dev device browser in VS Code.
- Right click on your brick and *Open SSH Terminal*
- Run this command:
```
sudo apt update && sudo apt install pybricks-micropython
```
- If prompted for a password, it is: *maker*
- And then it will take a while. It's usually the fastest with a WiFi dongle, or a USB-Ethernet converter if you have one

devinbreise, 2020-04-24:
Yes, on that particular bot, the team was highly focused on squaring the robot perpendicular to lines, so they have a color sensor placed directly in front of each drive wheel.  This isn't optimal for a line follow of course, but it is very similar to the educator design. 

With a "home cooked" PID controller and a little bit of coefficient tuning, I can start this bot on a straight line, with the color sensor directly over the white or black area (deliberately not centered on the line edge) and I will see some oscillation initially, then it will fade as the derivative smooths it out.  After a few inches, the robot is moving straight along the line edge with no visible oscillation.  

When I use the pybricks controller, "hooking up" the sensor through the angular velocity parameter to drive() (just as you did in the sample code), I can't get the oscillation to fade, it remains consistent as the robot moves along.

I'd be happy to try the default coefficients in the latest build.  I haven't put a wifi dongle on the EV3, just using a bluetooth connection back to a PC and using VSC and the ev3dev-browser plugin to move bits onto the brick.  You taught me how to "override" the pybricks library that is in the image by putting an alternative one in the project folder and adjusting the shebang.  Is there a way for me to download the latest lib version other than apt?

laurensvalk, 2020-04-24:
Fair enough - apt isn't very convenient - so here it goes: [pybricks-micropython-v2.0.0b6.zip](https://github.com/pybricks/support/files/4530633/pybricks-micropython-v2.0.0b6.zip)

I didn't change anything to the PID though, so I'm guessing this will not make a major difference.

Can you get it to work smoothly at low drive speeds and a low equivalent to `PROPORTIONAL_GAIN` in the example I shared?

Also, can you perhaps share the log file of the drive base?

```python

# init your robot here, let's say it's called my_drive_base

my_drive_base.log.start(30000)

# Insert code to follow a line for about 30 seconds

my_drive_base.log.save('drive_base_log.txt')
```

A file with that name should now appear in your project folder. It would be great if you could upload it here, and maybe share your exact code sample as well.

Then I'll run it through my script to create plots and share how you can do it as well if you like.

devinbreise, 2020-04-27:
With the 0424 build, and a robot whose light sensors are directly in front of its drive wheels (see attached photo), and using the left light sensor to track the line, I tried the following code on a straight line:
```Python
        self.drive.log.start(30000)
        timer.reset()
        i=0
        while not any(self.brick.buttons.pressed()):
            error = self.lightL.reflection() - 50 #high is ~90, low is ~10
            self.drive.drive(100, error * -1.2)
            i=i+1
        self.drive.stop()
        print ("time:"+str(timer.time())+" iters:"+str(i)+"loop time:"+str(timer.time()/i))
        self.drive.log.save('drive_base_log_defaults.txt')
```
Settings were defaults (adjusted by the DriveBase constructor for the specified wheel diameter of 50 and axle track of 99) as follows:
```
-----------------------------------------
DriveBase Settings: straight_speed=116 straight_acceleration=465 turn_rate=134 turn_acceleration=538
-----------------
distance_control:
Limits: speed=349 acceleration=1396 actuation=100
Stall: speed=7 time=200
PID: kp=200 ki=600 kd=2 i range=10 i rate=2 ff=0
-----------------
heading_control:
Limits: speed=404 acceleration=1616 actuation=100
Stall: speed=8 time=200
PID: kp=200 ki=600 kd=2 i range=11 i rate=3 ff=0
```
While the robot tracked the line successfully, the oscillation would consistently increase up to some limit and then stay there.  The log file for that test is 
[drive_base_log_defaults_1.2_straight.txt](https://github.com/pybricks/support/files/4541993/drive_base_log_defaults_1.2_straight.txt)

I then tried the same code on the line on this season's City Shaper mat that extends into the launch area and noted that the maximum error multiplied by a coeffecient of 1.2 could not produce a ratio of angular speed to forward speed sufficient for the robot to track the 45 degree turns on that line (at least the right turns).  Some testing determined that in order for this robot to successfully track the 45 degree turns on that line (without overshooting to a point where it became lost), the ratio of forward speed to angular speed needed, in the extreme case, to be around 1:1.  Given a default forward speed of 100, this means a proportional adjustment on the light sensor reading of 2.5 (max error on reading is +-40).  Adjusting the -1.2 constant in the code to -2.5, the robot successfully tracked that light through the 45 degree turns, but the oscillation increased and again, did not dampen on the straight sections.  The log file for that test is 
[drive_base_log_defaults_2.5_city_shaper.txt](https://github.com/pybricks/support/files/4542013/drive_base_log_defaults_2.5_city_shaper.txt)

Finally, I ran the robot on a straight line with the -2.5 constant and observed a similar, but larger steady oscillation.  The log of that is 
[drive_base_log_defaults_2.5_straight.txt](https://github.com/pybricks/support/files/4542020/drive_base_log_defaults_2.5_straight.txt)

Loop times on all tests remained < 2ms.

Looking forward to your thoughts!  Thx!
[image]

SQLSourcerer, 2020-11-30:
I took a look at the sample code, and I’m curious: what’s the 10ms wait for?

dlech, 2020-12-01:
I'm not sure which example code you mean, but usually we add a short wait in a loop so that the program doesn't use 100% CPU. On EV3 this is important since it is running the Linux operating system in the background and it needs CPU time too.

laurensvalk, 2023-03-17:
The limitations discussed here should be fixed now, so let's close this issue.

closed 2023-03-17 by laurensvalk