pybricks/support #20: Can't seem to "break" turn_acceleration

laurensvalk, 2020-04-21:
_Originally posted by @devinbreise in https://github.com/pybricks/support/issues/17#issuecomment-616871775_

> Testing with ca65b2f I can see that the bug mentioned above does seem to be fixed. Thanks! However, I am still seeing something I don't fully understand.
> 
> I'm testing turn() on RobotDrive using settings() with a set turn_velocity around 250 and adjusting the turn_acceleration parameter. My goal was to discover, for the robot I'm testing, at what threshold of turn_acceleration the turn() maneuver becomes inconsistent (due to wheel slippage). This is an experiment we take FLL teams through all the time, though usually with varying rates of velocity instead of acceleration because the (steep!) acceleration curves are fixed in the MindStorms blocks. Code looks like this:
> 
> ```python
>             self.drive.settings(turn_acceleration=maxAccel)
>             self.drive.turn(360)
>             self.drive.stop()
>             self.hold() #this calls hold() on the two drive motors
> ```
> 
> The call to hold() is to "lock" the motors in case there is any residual momentum.
> 
> What is surprising to me is that I can't seem to "break" it. As I increase the turn_acceleration value, I do see a shorter and shorter time to complete a turn(360) but it seems to cap out at a certain level, while it is still showing a high degree of accuracy. I thought perhaps the underlying Motor.control.limits were creating this cap, so I set the acceleration value there on both drive motors to a very high value, but that didn't seem to affect this cap.
> 
> So the good news is that the software is exhibiting a remarkable combination of speed/precision (360 degree spin in ~1.5 seconds with 1-2 degree accuracy with a medium sized robot). Very impressive.
> 
> But I feel there is still something in the software prohibiting me from getting it to the point where it loses accuracy. Perhaps the pid settings on the two wheels?

laurensvalk, 2020-04-21:
> So the good news is that the software is exhibiting a remarkable combination of speed/precision (360 degree spin in ~1.5 seconds with 1-2 degree accuracy with a medium sized robot). Very impressive.

That is good to hear.

> But I feel there is still something in the software prohibiting me from getting it to the point where it loses accuracy.

Right, they are still capped by the actual acceleration parameters in `distance_control` and `heading_control`. We need to work on either making this more intuitive or just documenting it precisely.

For one thing, we make sure that when you print `settings()` after you've exceeded the limits, at least we should show you that they have been capped.

Basically, `distance_control` and `heading_control` are the only "real" settings here. The ones in `settings` are just lower values that get set temporarily during `turn` and `straight`.

An even simpler solution might be to make them the same, but then the default acceleration would have to be pretty high to ensure line following works. You do say that this still leaves it pretty accurate, so maybe it's possible.

devinbreise, 2020-04-21:
The light is slowly dawning for me.  Here is what I think are the full collection of settings that are potentially relevant for DriveBase (B is left_motor, C is right_motor) with their default values in ca65b2f

```
-----------------------------------------
Motor B Settings:
Limits: speed=800 acceleration=1600 actuation=100
Stall: speed=15 time=200
PID: kp=400 ki=1200 kd=5 i range=23 i rate=5 ff=0
-----------------------------------------
Motor C Settings:
Limits: speed=800 acceleration=1600 actuation=100
Stall: speed=15 time=200
PID: kp=400 ki=1200 kd=5 i range=23 i rate=5 ff=0
-----------------------------------------
DriveBase Settings: straight_speed=100 straight_acceleration=200 turn_rate=200 turn_acceleration=90
-----------------
distance_control:
Limits: speed=349 acceleration=698 actuation=100
Stall: speed=7 time=200
PID: kp=200 ki=600 kd=2 i range=10 i rate=2 ff=0
-----------------
heading_control:
Limits: speed=396 acceleration=792 actuation=100
Stall: speed=7 time=200
PID: kp=200 ki=600 kd=2 i range=11 i rate=2 ff=0
```

If I understand:
1) distance_control and heading_control settings are derived at the point the DriveBase is configured by looking at the current left_motor and right_motor control settings, and applying the appropriate math given the specified axle_track and wheel_diameter.  From that point on, the control values on the two motors are irrelevant for all DriveBase methods and will, in fact, be ignored if changed.
2) DriveBase settings() fall into two categories.  Straight_speed and turn_rate are the only place where the desired maximum velocity for straight() and turn() can be specified.  Straight_acceleration and turn_acceleration, on the other hand, are a sort of override of the acceleration parameter on the drive’s distance_control and heading_control but only for the straight() and turn() methods, not the drive() method.  In addition, these overrides can further limit acceleration, but can not increase it beyond the drive’s distance_control and heading_control settings.
3) If the above is correct, there still remains confusion on my part as to the relationship between the control.limits.acceleration setting and the control.pid settings.  Is one intended to cap the behavior of the other?  Or provide a simpler way of configuration for the desired behavior?

Disregarding 3 for a moment and presuming 1 and 2 above are correct, if I want to find thresholds where the combination of commanded velocity and maximum allowed acceleration in straight() and turn() cause accuracy/consistency to fall off (due to wheel slippage), I would need to adjust DriveBase.distance_control.limits.acceleration and DriveBase.heading_control.limits.acceleration to arbitrarily high numbers, and then use DriveBase.settings to set the relevant velocity and max acceleration for sequential tests.

In looking at my previous code, I see that I set limits.acceleration on the two underlying motors to that arbitrarily high value, but didn’t adjust the DriveBase.heading_control.acceleration setting.  That was apparently my mistake.

Modifying the code to only adjust the DriveBase.heading/distance_control.acceleration settings (leaving the Motor.limits.acceleration values at the defaults), I am now seeing the behavior I expected and can find thresholds through trial and error.

laurensvalk, 2020-04-22:
Yes, that's correct. Here's a bit of background as to why, which might help with the understanding. We'll want to document this as well.

**There is only one type of *PID-like* controller in Pybricks.**

It ensures that a measured "position" output tracks a given "reference position" and "reference speed" pattern. Usually, it is a position going from A to B or a speed going from C to D.

I put "position" and "speed" in quotes: the generic controller just controls "counts" and "rate" of change of the count value. Usually, a "count" is an encoder pulse coming from the motor. These counts are scaled by a constant factor to give the physical units. Try printing `mtr.control.scale` or `robot.distance_control.scale`.

For example, EV3 motors have 2 counts per degree, so `scale=2`. Try it with `gears=[12, 36]`. Now the scale is `2*3=6`.

And all these controllers are the same. Their settings are changed through the control methods.

**A `Motor` has one controller. A `DriveBase` has one for distance, one for heading**.

Aside from rather interesting implementation details that make it more accurate and responsive, it really still boils down to PID control at the core. For a single motor, this is fairly commonly understood.

For the `DriveBase` there are two PID controllers. However, *not* one per motor. So they don't control `count_left` and `count_right` but they control `(count_left + count_right)` and `(count_left - count_right)`.

With a bit of geometry, you'll find that these are just one `scale` factor away from distance and heading.

But as far as the control implementation and control settings go, it's all the same.

The only exception is the `.settings()` method, which is just used to temporarily adjust `distance_control` and `heading_control` to improve the experience with `straight` and `turn`, so that beginning users don't have to mess with settings.

laurensvalk, 2022-08-23:
This should be fixed by now. The acceleration setters now propagate to the underlying control settings.

closed 2022-08-23 by laurensvalk