pybricks/support #22: Writing in a pre-existing file

DLunomos, 2020-04-26:
I'm currently doing a project that involves calibrating a older color sensor that doesn't have support for it. I'm planning to store the raw data in a csv file. Pybricks does support data logging, but you must create a new file every time the program runs. If you could give the log or DataLog class an argument that states if it's writing in a pre-existing file, it would be a lot simpler to store larger amounts of data.

dlech, 2020-04-26:
Do you mean that you would like to append additional data to existing data already in the file without overwriting the existing data?

DLunomos, 2020-04-26:
Yes, that's what I intended.

laurensvalk, 2020-04-26:
Thanks @DLunomos  for your suggestion. I've added experimental support for it earlier this morning, which might make it into the next release.

laurensvalk, 2020-04-27:
Support added in https://github.com/pybricks/pybricks-micropython/commit/8097a97ff70d99e3b0e356aebd2e6b624b7c60f5

This will be included in the next release in a week or so.

There is now an extra keyword argument `append` when you create the `DataLog`. It is `False` by default so existing code is unaffected.

I'll keep this issue open until it's documented as well.

laurensvalk, 2020-04-27:
Documentation added in https://github.com/pybricks/pybricks-api/commit/9ff69a54fb404c54c4576a9264780b7fdbf5cd5f.

Thank you @DLunomos!

closed 2020-04-27 by laurensvalk