pybricks/support #23: Multitasking support

CharlesTam20, 2020-04-26:
**Question**
I was wondering whether there's support on multitasking.

**Context**
Independent unit of task similar to EV3-G.

**Screenshots**
NA

dlech, 2020-04-30:
The Pybricks library is not thread-safe, so we try to avoid using threads.

Instead we recommend using generator functions as coroutines. The [Gyro Boy example program][ex] makes use of this.

[ex]: https://github.com/pybricks/pybricks-projects/blob/51871031b8bb93dc1f69ba99a97536c98959e622/official_models/ev3/education_core/gyro_boy/main.py#L82-L173

In v2.0 there are some operations, like playing sounds, that don't have a non-blocking version, so unfortunately can't be used in coroutines. In that case threads are the only option. https://github.com/pybricks/support/issues/7#issuecomment-607974144 has an example of how to add locks to make playing sounds thread-safe.

CharlesTam20, 2020-04-30:
Hi David, 

Thank you for your reply & suggestion.
I’ve been testing the Beta version of Pybrick Micropython; it’s quite speedy when compared with the Python3 version.
However, Python3 does offer good support on Threading. 
I’ll try to use the “generator functions as coroutine” with Pybrick.

closed 2020-04-30 by CharlesTam20