pybricks/support #26: [Question] Hard Reset for Gyro drifting?

johnscary-ev3, 2020-04-28:
I am using the Gyro to do angle controlled turns.
There is a section of code below of 90dg Left Tank turn that can use Gyro if Available.
Mainly this works fine and is not sensitive to slow Drifting of the Gyro since I reset its angle to 0 just before each turn.
However I have noticed that sometimes the Gyro Drift rate suddenly increases  and it become not usable. Drift rate seems to increase over time or can be high from the start if you move the Gyro during start up of the system. Only way I have found to fix this is power off/on reset of system.
So, is there a way to do the equivalent of the power off/on reset with software command?
Thanks.

```
#90dg Left tank turn using gyro if available
def left_tank_turn(BC_turn_speed, motorB, motorC):
    global GyroAvailable
    global Rotation_angle_90dg_tank_turn
    if GyroAvailable:
        gyro.reset_angle(0)
        while gyro.angle() != 0:
            gyro.reset_angle(0)
            wait(50)

        motorB.run(BC_turn_speed/2)
        motorC.run(-BC_turn_speed/2)

        while gyro.angle() > -89:
            wait(10)
        
        motorB.stop(Stop.BRAKE)
        motorC.stop(Stop.BRAKE)
    else:
         motorB.run_angle( BC_turn_speed/2, Rotation_angle_90dg_tank_turn, Stop.BRAKE, False)
         motorC.run_angle( BC_turn_speed/2,-Rotation_angle_90dg_tank_turn, Stop.BRAKE, True)

```

johnscary-ev3, 2020-04-29:
Update:
  While I have seen large Gyro drifting, I found in this case, (by putting a print statement in the loop), the Gyro suddenly starts to return all zero readings. So then the tank turn does not stop.
Power off/on will fix the Gyro in this case also.
 Any idea on what may cause this?
Thanks
```
#90dg Left tank turn using gyro if available
def left_tank_turn(BC_turn_speed, motorB, motorC):
    global GyroAvailable
    global Rotation_angle_90dg_tank_turn
    global motion_print
    if GyroAvailable:
        gyro.reset_angle(0)
        while gyro.angle() != 0:
            gyro.reset_angle(0)
            wait(50)

        motorB.run(BC_turn_speed/2)
        motorC.run(-BC_turn_speed/2)

        angle=gyro.angle()
        while angle > -89:
            if motion_print:
                print("LT_Angle= ",'{:.2f} '.format(angle) )
            wait(10)
            angle=gyro.angle()
        
        motorB.stop(Stop.BRAKE)
        motorC.stop(Stop.BRAKE)
    else:
         motorB.run_angle( BC_turn_speed/2, Rotation_angle_90dg_tank_turn, Stop.BRAKE, False)
         motorC.run_angle( BC_turn_speed/2,-Rotation_angle_90dg_tank_turn, Stop.BRAKE, True)
```

laurensvalk, 2020-04-29:
Since this looks like part of a program, do you also use `gyro.speed()` anywhere in your program? 

(There is a known hardware problem that prevents you from using both in the same program, which we haven't documented yet.)

> Only way I have found to fix this is power off/on reset of system.

You can also unplug the cable, which is faster than rebooting the whole EV3. Be sure to hold the sensor absolutely still while you do.

> So, is there a way to do the equivalent of the power off/on reset with software command?

Unfortunately, the hardware issue kind of prevents a solution that is stable enough for us to add officially.

As a hack, you can do `AnalogSensor(Port.S1)` on that port (change S1 to the one you use) before you initialize the Gyro. This forces that sensor port to reboot. But sometimes you have to restart your script if it doesn't work at once, so it's not perfect. Still, it's usually easier than rebooting the whole EV3 or unplugging a cable.

johnscary-ev3, 2020-04-29:
ROBODOZ3R EV3 micro-python-R3 JS 4/24/2020
Traceback (most recent call last):
  File "/home/robot/ROBODOZ3R_micro_py_R3/main.py", line 162, in <module>
  File "/home/robot/ROBODOZ3R_micro_py_R3/motion_tools.py", line 41, in Init_Gyro
OSError: [Errno 19] ENODEV: 

A sensor or motor is not connected to the specified port:
--> Check the cables to each motor and sensor.
--> Check the port settings in your script.
--> Check the line in your script that matches
    the line number given in the 'Traceback' above.

closed 2020-04-29 by johnscary-ev3

johnscary-ev3, 2020-04-29:
Hi,
Thanks for the help.  Not using gyro.speed().
I found AnalogSensor in pybricks.iodevices 
Hope that is correct.
When I use it, as in code fragment below, I get this error about half the time.
Should I do some wait between calls or code it differently?

```
ROBODOZ3R EV3 micro-python-R3 JS 4/24/2020
Traceback (most recent call last):
  File "/home/robot/ROBODOZ3R_micro_py_R3/main.py", line 162, in <module>
  File "/home/robot/ROBODOZ3R_micro_py_R3/motion_tools.py", line 41, in Init_Gyro
OSError: [Errno 19] ENODEV: 

A sensor or motor is not connected to the specified port:
--> Check the cables to each motor and sensor.
--> Check the port settings in your script.
--> Check the line in your script that matches
    the line number given in the 'Traceback' above.

from pybricks.iodevices import ( AnalogSensor)
# Initialize Gyro
def Init_Gyro():
    global GyroAvailable
    GyroAvailable = True
    global gyro
    AnalogSensor(GyroPort)
    gyro = GyroSensor(GyroPort)
    gyro.reset_angle(0)
```

dlech, 2020-04-29:
> Should I do some wait between calls or code it differently?

Yes, adding a wait should help. It can take several seconds for the gyro to reboot and sync back up. 

Or add a retry:

```python

retry = 2
while retry:
    try:
        gyro = GyroSensor(Port.S1)
        break
    except OSError:
        retry -= 1
```

laurensvalk, 2020-04-30:
> Yes, adding a wait should help. It can take several seconds for the gyro to reboot and sync back up

No, that does not help. The code doesn't work that way.

Internally, the code already tries up to 6 times, waiting 1, 2, 4, 8, and finally 16 seconds between each attempt.

Sometimes, it just never comes back. This is why we can't add it to the main code and it stays a hack.

Retrying and waiting even longer than 30 seconds *sometimes* helps, but still not reliably enough to work all the time.

The same is true for the other LEGO UART sensors, but fortunately we don't need to reset those.

johnscary-ev3, 2020-05-04:
Actually I found that the problems I saw with the Gyro are due to a bad connection from brick to gyro. The connection is ok at start of program at Initialize, but apparently goes intermittent while robot is moving around. Then the gyro starts returning all zeros and my gyro turn can fail.
I am wondering why the system does not return some sort of error when this happens or can I do some sort of check to make sure communication is still ok?
Should I just Initialize it before each use? Would that help?