pybricks/support #27: [Question] parallel code execution

AchimTack, 2020-05-03:
**Preamble**
Hi everyone! I just discovered this great project and I'm thinking about switching to pybricks for my ffl-team.

**Question**
Has anyone tried to to parallelize code execution with pybricks? I tried to use multiprocessing but until now with no proper results. Maybe someone can give me a hint on how to accomplish this?

**Context**
My use case is that I want to get a tool motor in the right position while approaching a target at the same time.

laurensvalk, 2020-05-03:
Thanks for sharing the context as well. This is a good case that you might be able to do nicely without parallel code.

```python
my_motor.run_target(500, 720, wait=False)

print("The program continues while the motor is still busy going to its target angle.")
print("So you see this text even though the motor is still moving.")

```

The trick is the `wait=False` parameter. Does this help?

AchimTack, 2020-05-03:
Thanks for your quick response!
Unfortunately I consistently get the following error message:
[image]
Sorry, this might be a complete noob question / problem :/

laurensvalk, 2020-05-03:
Ah, yes, keyword arguments were introduced in version 2.0. Since you say you are just beginning with Pybricks, I recommend that you start with that version. It’s technically still a beta, but nearly final. See pybricks.com/downloads.

If you prefer to use version 1.0, you can still use it, but you have to supply all the optional arguments:

ml.run_target(500, 720, Stop.HOLD, False)

AchimTack, 2020-05-03:
Thanks for your fast reply! I'll upgrade to 2.0 then. Just for clarification: Is the documentation still based on 1.0 or 2.0?

laurensvalk, 2020-05-03:
docs.pybricks.com is all 2.0.

The version on the official LEGO site is still 1.0, but hopefully they will upgrade soon as well.

We’re working on an upgrade guide to make clear what is what.

devinbreise, 2020-05-04:
I'll add that I did a few tests with threading for similar reasons, but needing more complex series of operations to proceed in parallel (e.g. stall a medium motor, reset the angle, rotate to a specific angle) with robot movement.  I wrapped the parallel code up in a function definition and then included these lines in the "main" thread:
```Python
        threading.Thread(target=self.resetTower).start() #run resetTower() in another thread
        threading.Thread(target=self.resetBasket).start() #run resetBasket() in another thread
```
And all seemed to work quite well with the other 2 threads reaching the end of the function, signaling that those threads were done and could be recycled.

AchimTack, 2020-05-07:
@devinbreise sounds promising! I'll definitely try this out.

DietrichChristopeit, 2021-06-17:
I had very good results using asyncio.
For me that was then the way to go.

dlech, 2023-03-21:
Future visitors will be interested in the latest developments #917.

closed 2023-03-21 by dlech