pybricks/support #28: [Question]  How to connect IMU sensor from mindsensors.com

Sunrise98, 2020-05-04:
**Question**
I want to use Absolute0-IMU sensor from mindsensors.com on pybricks.
I tried to use it using Ev3dev class, but "RuntimeError:Unknown Error" was occured many times.
Sometimes, the connection was successful and get values when I reconnect the cable to the sensor port.
Please tell me how to connect the sensor correctly.

**Context**
The codes I tried is here. The compass sensor from Hi-Technic could get the value normally, but IMU sensor from mindsensors.com cloud not connect. 
```
from pybricks.iodevices import Ev3devSensor 

sensor = Ev3devSensor(Port.S1)
while True:
    val = sensor.read('COMPASS')
    print(val[0])
    wait(500)
```

laurensvalk, 2020-05-04:
Are you able to see the sensor values in the Device Browser from the main menu on the EV3 brick?

Sunrise98, 2020-05-04:
Yes, I can see the value on Device Browser ("ms-absolute-imu" and "Mode:COMPASS" is selected, and I can watch the value of compass), but "Runtime Error: Unknown Error" was occurred in instance initialization of Ev3devSensor.

laurensvalk, 2020-05-04:
Thanks for reporting!

laurensvalk, 2020-05-04:
The data type for that sensor mode was not yet supported. I just added support. This will come in the next release - thanks for finding this.

Sunrise98, 2020-05-04:
Thank you for your quick response.
I'm looking forward to the bug fix and using the sensor on pybricks!

laurensvalk, 2020-05-20:
The latest version is [out now](https://pybricks.com/downloads/), so I think we're ready to close this. Thanks!

closed 2020-05-20 by laurensvalk