pybricks/support #29: Roadmap for next release

laurensvalk, 2020-05-07:
**Pybricks 1.0 was all about bringing Pybricks MicroPython to EV3.**

This came out early 2019. It's super exciting to see people using this actively.

An epic number of users! According to [Visual Studio Code, anyway](https://marketplace.visualstudio.com/items?itemName=lego-education.ev3-micropython).

If you're one of those, click :+1: on this post so we know you're here! :wave: 

**Pybricks 2.0 was all about improving the EV3 experience.**

To make it absolutely solid, tested, fully-featured, and FLL-ready. 

It's ready now (:rocket: :tada: ), and should be available in the latest ev3dev build soon.

Hopefully LEGO will update their official derivative version soon too.

**Pybricks 3.0 will add support for Powered Up!**

- Newly supported hubs / programmable :bricks: 
   - LEGO BOOST MoveHub: ``MoveHub()``
   - LEGO Powered Up City Hub: ``CityHub()``
   - LEGO Technic Control+ Hub: ``CPlusHub()``
- Newly supported motors / sensors:
   - BOOST Color and Distance Sensor
   - BOOST Interactive Motor
   - Technic Control+ Motors (L, XL)
   - SPIKE Prime Motors (M, L)
   - SPIKE Prime Sensors (Color/Ultrasonic/Force)
- Go fully open source :nerd_face: 
- **Pybricks Code**: Cross-platform editor for Powered Up. No install required!
   - Lets you install Pybricks MicroPython firmware
   - Lets you run Pybricks MicroPython scripts
   - Windows :desktop_computer: / Mac :computer: / Linux :penguin:  / Android :phone: / ChromeOS :mag: 
- By default, a script runs from memory:
   - This is super quick and easy :rocket:, but you have to be connected to start the program :zap: .
   - Use this to test & develop your script.
- Optionally, a script can be included when you update the firmware:
  - This is slow to download :snail:, but your script stays :tada:!
  - Use this when you're happy with your program (or if you have unlimited patience)
  - Start your script with the button. No Bluetooth connection required. Take your hub anywhere!
  - You have to send us footage if you send it into space :rocket: 

**Pybricks 4.0/5.0 will be ... ?**

In other words, what ***might*** we do one day, in a galaxy far away, ***but certainly not right now***?

This is all super cool too, but it's better to stay focused on making the essentials really good first.

- Support for internal sensors like accelerometer and IMU.
- Support for older (NXT) and newer hubs (Mario)
- Support for LEGO SPIKE Prime
- Powered Up hub-to-hub communication
- Powered Up remote handset support
- Improvements to EV3 Visual Studio Code extension

There are also a few things that are just not possible due to hardware limitations:
- Support for WeDo 2.0: Unless we missed something, this firmware can't be updated. So you can't install Pybricks.
- Most of the hubs have sufficient flash and RAM capability for Pybricks. But the BOOST Move Hub has less than half of what the others have. So, not all new features beyond 3.0 will come to the BOOST Move Hub.

Other than that, the sky is the limit!

GianCann, 2020-05-07:
Great work!!!

phipsitn, 2020-05-07:
Will having coordinated simulator be part of the roadmap?  There is another github project called ev3dev-sim where I was trying to make it pybricks 1.0 compatible but got stuck part way in.  Having a way to let FLL team members test navigation code before putting it on the robot is really helpful.  Maybe this would be a good project to bring into the fold?

ws1088, 2020-06-10:
I would love to see support for Spike Prime Hub. Currently the current LEGO Python API is really lacking.

johnscary-ev3, 2020-06-15:
Will Pybricks 3.0 support the new MINDSTORMS Robot Inventor Hub just announced by LEGO on 6/12/20?
The LEGO release document mentions Python support.
Thanks.

laurensvalk, 2020-06-15:
From the pictures, the MINDSTORMS Robot `InventorHub` looks very similar to the SPIKE `PrimeHub`.

Hopefully, the electronics are 100% the same instead of 99% the same, so we only have to make one firmware :smile:

Please see the note about SPIKE Prime in the note above. I'll update the post, thanks!

johnscary-ev3, 2020-06-16:
Yes it looks just like SPIKE Prime except for the case colors ;0)
I notice you have SPIKE Prime and MINDSTORMS Robot Inventor under Pybricks 4.0/5.0 above.
Hopefully it goes ok.
Thanks.

Nikolay-Zagrebin, 2020-06-27:
any plans for RCX?

laurensvalk, 2020-06-27:
> any plans for RCX?

Our adaptation of MicroPython is very compact, but I’m afraid 32 KB just isn’t going to work 😄 

However, with some work you can probably use the USB IR Tower with Pybricks for EV3. People have done this with ev3dev already I think. This is not something we will officially support or develop, but it would certainly be fun if someone in the community wants to look at this and create a nice example project.

TheVinhLuong102, 2020-09-02:
@laurensvalk @dlech may I ask if you plan to add support NXT later this year or early 2021?

The main rationale for my question is that I'm currently working on a non-profit that is aiming to help tens of thousands of school children in a developing country to get exposure to robotics programming. Even with financial donations, we're priced out of the EV3 at this scale. I'm thinking the NXT sets, which are currently costing only 30% as much as the EV3 sets, will be an appropriate medium, if students still get to program the NXT-based bots with state-of-the-art APIs such as PyBricks.

laurensvalk, 2020-09-02:
We would ideally like to add NXT support eventually, for the very same reasons you mention. It would be very cool to give all those unused NXT sets a new life. We have already shown that it is technically possible. Unfortunately, we can't prioritize it at this time as you can see in the roadmap. If we ever do it, we may have to find some sponsors to help make it happen.

TheVinhLuong102, 2020-09-02:
@laurensvalk, it's great that you already have the very same idea! The NXT hubs, which are actually quite powerful hardware, will be wasted without a modern, preferably Python, API. If I'm not wrong, the existing Python API for NXT is https://github.com/Eelviny/nxt-python, which is about 5 years old already and currently unmaintained.

What does it take, generally, to write the NXT hub module in PyBricks? How can we help? -- I can look among my friend circle for engineers who would like to take on this, with your guidance. Also, what sponsorship is needed for projects like this?

dlech, 2021-04-02:
It's been almost a year in the making, but we are getting very close to an official v3.0 release :rocket: :robot: 

We will start a new discussion for the v3.1 in GitHub Discussions.

closed 2021-04-02 by dlech

petitlen, 2021-04-05:
> From the pictures, the MINDSTORMS Robot `InventorHub` looks very similar to the SPIKE `PrimeHub`.
> 
> Hopefully, the electronics are 100% the same instead of 99% the same, so we only have to make one firmware 😄
> 
> Please see the note about SPIKE Prime in the note above. I'll update the post, thanks!

How long do you release to support Lego MindStorm Inventor