pybricks/support #30: [Question] Wifi support?

johnscary-ev3, 2020-05-14:
**Question**
I understand I need to use a USB Wifi dongle to get Wifi access with pybricks.
NetGear WNA1100 Wireless-N 150 seems to be the one supported from ev3dev site but can we use a newer and smaller NetGear dongle?

https://www.ev3dev.org/docs/networking

**Context**
I have been using Bluetooth for my wireless connection but it seems to have very limited range on two PCs I have tried and drops the connection when robot is moving around. 
So I am thinking Wifi would be better. Is that correct?

**Screenshots**

dlech, 2020-05-14:
Bluetooth should typically have a range of 10 meters (33 feet) not accounting for obstructions and radio interference.

There is more information on ev3dev and Wi-Fi dongles at https://github.com/ev3dev/ev3dev/wiki/USB-Wi-Fi-Dongles

johnscary-ev3, 2020-05-14:
Ok, thanks for the info.
Not sure why I don't get more than about 10 feet on my Bluetooth.
It works on my desk but when I put robot on the floor in same room it will drop off.
I am going to try the Edimax EW-7811Un for wifi since it is very small and inexpensive.
Found it on Amazon.

johnscary-ev3, 2020-06-01:
Hi,
I got the Edimax wifi module and it works great. Small size is nice so it does not hit anything.
Just need to disable the Bluetooth or the system keeps trying to use that when you reboot it.
Thanks.

laurensvalk, 2020-10-09:
This is resolved. Thanks for your question!

closed 2020-10-09 by laurensvalk