pybricks/support #31: [Question] Simulator for Pybricks robots

devinbreise, 2020-05-21:
I'd like to start a thread here to gather knowledge about simulators that might be used with Python (ideally Pybricks) based robots.  Many FLL teams are, of course, not meeting physically right now, and their coaches are looking for ways to keep the learning going.

This one (https://www.aposteriori.com.sg/Ev3devSim/index.html) is quite simple, doesn't do much with physics, but does get the job done.  It supports programming against the Python EV3DEV API.  Maybe someone wants to do the Pybricks version?

These folks in Canada (https://www.virtualroboticstoolkit.com/) have gone much further, but currently don't support Python.  However, they tell me it is on their road map "soon".  Maybe they would partner with Pybricks to produce that.

What else are folks aware of in this space?

laurensvalk, 2020-05-22:
> Will having coordinated simulator be part of the roadmap?  There is another github project called ev3dev-sim where I was trying to make it pybricks 1.0 compatible but got stuck part way in.  Having a way to let FLL team members test navigation code before putting it on the robot is really helpful.  Maybe this would be a good project to bring into the fold?

_Originally posted by @phipsitn in https://github.com/pybricks/support/issues/29#issuecomment-625483812_

laurensvalk, 2020-05-22:
Although simulations are not currently on our [roapmap](https://github.com/pybricks/support/issues/29), we may consider simulations for other use cases as well. It would be quite useful to systematically and deterministically debug and unit-test our motor and sensor drivers. That level of detail is not necessarily useful for the goal of learn robotics and programming though.

XilefTech, 2020-09-14:
I think for the first part, something like a version of the pybricks software that you could run in some sort of vm would be nice where u got the display, can connect to that virtual robot to VScode and upload/start code to test for errors like Typos and stuff. That sould not be much work I guess but would help a lot when trying to just test, wether everything is okay in the code when working without an ev3 (or other hub) nearby

laurensvalk, 2022-08-23:
Just to follow up: we did make some real progress in this direction.

There is now a virtual hub that we use to test Pybricks on the CI. That's not immediately useful for the application discussed here, but it means we now know what it might take to make this happen if there is sufficient interest.

> These folks in Canada (https://www.virtualroboticstoolkit.com/) have gone much further, but currently don't support Python. However, they tell me it is on their road map "soon". Maybe they would partner with Pybricks to produce that.

We're always happy to talk to third parties who might be interested in using Pybricks. We can be reached at team@pybricks.com