pybricks/support #32: [Question] Every time I try to burn the new 2.0 Disk Image to my microSD, I get an error

anttesoriero, 2020-05-27:
**Question**
When I use balenaEtcher to burn the new stable 2.0 release to my microSD, I get an error about the image. Do you know why this would be happening? I've used the same method for other updates before and it was always fine.

**Context**
Install the new stable 2.0 release on my microSD.

**Screenshots**
[image]

dlech, 2020-05-27:
What is the URL you used to download the SD card image?

anttesoriero, 2020-05-27:
https://pybricks.com/downloads/

dlech, 2020-05-27:
Have you tried downloading it again?

anttesoriero, 2020-05-27:
Yes, but I figured it out kind of. For some reason it's only an issue on balenaEtcher. When I tried another application, it worked. I'm using a Mac, so I used ApplePiBaker instead.

dlech, 2020-05-27:
Which version of balenaEtcher?

anttesoriero, 2020-05-27:
Version 1.5.56 (1.5.56)

laurensvalk, 2020-10-09:
Is this still an issue, @anttesoriero?

closed 2021-07-13 by laurensvalk