pybricks/support #33: [Question] Soudfile playback without waiting for end of soundfile

XilefTech, 2020-05-31:
**Question**
Is it possible to play a sound file while doing something different?

**Context**
I want to make a little menu on my brick that you can navigate through with the buttons. I want to play a click sound each time, a button is pressed. But in this case playing the sound file takes enough time to add unnecessary delay. 
The documentation only tells about speaker.play_file(soundfile), but is there something like speaker.play_file(soundfile, awaitEnd) ?
What would be your solution to this?

laurensvalk, 2020-05-31:
Could a short, low frequency beep be good enough to mimic a click sound? It starts a lot faster, reducing delay.

XilefTech, 2020-05-31:
Well it works for the moment but I'd like to use sound files for that in the future.
Could it be possible, to use the treading lib to create a separate thread that play the sound while my mainthread does the other stuff?

dlech, 2020-05-31:
Please see https://github.com/pybricks/support/issues/7#issuecomment-607974144

XilefTech, 2020-05-31:
Okay, Thanks!

XilefTech, 2020-05-31:
okay, I now have made it with a thread that plays the sound. The loading times of the sound before it actually starts playing still seem to be a bit high tho. 
There may be some room for improvement for the Future ;) 
Anyways, thanks for the help!

closed 2020-05-31 by XilefTech

dlech, 2020-05-31:
Are you using one of the sounds from `pybricks.media.ev3dev` or your own? If it is your own, what is the file format (bit rate, compression, etc.)?

XilefTech, 2020-05-31:
I used both sounds from pybricks.media.ev3dev and a custom .wav with a sampling rate of 44100 and a bit depth of 16 bit