pybricks/support #34: Run button not active

Patlepirate01, 2020-06-07:
**Question**
The run button is never active after the bluetooth activation (and the bluetooth button is not active again but I don't know if it's normal or not) ==> not sure if the bluetooth connection is successfull or if another issue after

**Context**
I'm using a macbook air with Mac OS Sierra (10.13.6) and Chrome 80 (as requested). The flashing of the hub was successfull (when I press after the green button one motor is turning very shortly). Then I try to connect Bluetooth. Icon is changing on chrome (Tab associated with a bluetooth device) but the Run button is not active

**Screenshots**
[image]

Patlepirate01, 2020-06-07:
ANd I have also tested on my imac with Catalina (10.15.5) and Google 85 and I see the device well connected in the bluetooth list but still no ccess to the run button
[image]

laurensvalk, 2020-06-07:
Thanks for the detailed report @Patlepirate01 ! Here are two things you can try.

### Refreshing the pybricks code cache:
Sometimes, a normal refresh not enough. You may have to clear the cache also. Here's how you can do it on Windows/Mac.

[image]

### Deleting the bluetooth device
1. Close your browser.
2. In your Bluetooth settings on your Mac, delete the Pybricks Hub

Can you try again after following these steps?

Patlepirate01, 2020-06-07:
Thanks for the quick answer but unfortunately after many trials it's not working.  It looks like after the Bluetooth connection something is "blocked".  As you can see in the printscreen below: device seems to be well disconnected and reconnected (in the Bluetooth Mac settings at least) 
[image]
[image]
[image]

laurensvalk, 2020-06-07:
When you go to `help>about Google Chrome` as shown below, you which version do you see?

[image]

laurensvalk, 2020-06-07:
Also, could you try the following?

1. In Google Chrome go to chrome://flags/.
2. Search for `Experimental Web Platform features` and enable it.

Patlepirate01, 2020-06-07:
The Experimental Web feature was already enabled and my Google version is
[image]

Patlepirate01, 2020-06-07:
Forgot to say that I systematically test on my 2 macs (one Macbook Air with OS version 10.13  and one iMac with os version 10.15...)

phipsitn, 2020-06-07:
How does one get access to the code.pybricks.com system to help beta test?

laurensvalk, 2020-06-07:
> How does one get access to the code.pybricks.com system to help beta test?

Thanks for your interest! Check out the beta notes & contact here: https://docs.pybricks.com/en/latest/start_pup.html

dlech, 2020-06-07:
I am able to reproduce the problem. Flashing firmware works but connecting to the NRF UART service after that seems to hang forever without connecting. No errors appear until device is disconnected in which case we get the expected disconnected while retrieving services error.

dlech, 2020-06-08:
Hi Pat,

I just uploaded a new firmware with a fix. So reload the app using the instructions above and flash the firmware again. Then connecting Bluetooth should work on macOS.

closed 2020-06-08 by Patlepirate01