pybricks/support #35: Comparing ColorSensor.color variable with string

Munchkin255, 2020-06-07:
Thank you for creating pybricks for the EV3. I am really pleased by the possibilities of writing code in Python.

**Question**
I am trying to use the EV3 Color Sensor for navigation. But I cannot figure out how I should write my if-statement based on the color output from the sensor.

**Context**
I have placed my robot on white piece of paper, and the color sensor is detecting a white color.
This is verified by print (str(colorsensor.color())) which outputs: Color.WHITE
However if I try to write an if-statement: if colorsensor.color() == "Color.WHITE": it evaluates as false.
What am I doing wrong?

[image]

laurensvalk, 2020-06-07:
Hi Carsten:

You can do it like this, no need for strings:

```python
colorsensor = ColorSensor(Port.S2)
if colorsensor.color() == Color.WHITE:
    print("Hello!")
```

laurensvalk, 2020-06-07:
And as an extra tip: In Python you can use the `print` function for all kinds of things - no need to convert things to strings first.

For example, you can do all of the following:

```python
print(12345)
print(1, 2, 3, 4, 5)
print(4, "apples and", 2, "oranges")
print([i**2 for i in range(10)])
print(colorsensor.color())
```

Munchkin255, 2020-06-07:
Thanks a million for both of your replies.

I notice, that I have been using print in a less than optimal way.
My code in order to combine the value of a variable with a label.
`print ("variable label" + str(variable)` 
I should have been using:
`print("variable label" , variable)`

Cheers Carsten

laurensvalk, 2020-06-07:
You're welcome - glad to hear you got it working!

closed 2020-06-07 by laurensvalk