pybricks/support #36: Using print() to track values causes connection with CPlusHub to fail

AlbanT, 2020-06-07:
I uploaded the demo program (for the record, the first time I tried it pybricks complained that my motors weren't connected to the hub, after reflashing it worked):

    # Initialize hub and sensor
    hub = CPlusHub()
    
    # We'll use two motors. One is a dial to
    # set the speed of the other motor.
    motor = Motor(Port.C)
    dial = Motor(Port.A)
    
    # Say hello :)
    print("Hello, Pybricks!")

    # First, we'll move the dial to absolute zero.
    dial.run_target(500, 0, Stop.COAST)
    
    while True:
          # Set the speed based on dial angle
         speed = dial.angle()*3
    
         if abs(speed) < 100:
             speed = 0
         motor.run(speed)
The single `print("Hello, Pybricks!")` works fine. But I tried to print the value of speed inside the while loop:

    while True:
          # Set the speed based on dial angle
         speed = dial.angle()*3
    
         if abs(speed) < 100:
             speed = 0
         print(speed)
         motor.run(speed)

This also works but when I press the stop button the hub doesn't stop and both the start and stop button are greyed out. I have to remove the batteries and reconnect to take control again.

I suspect the loop is too fast, when I insert a `wait(10` to reduce the speed of the loop it works fine:

    while True:
          # Set the speed based on dial angle
         speed = dial.angle()*3
    
         if abs(speed) < 100:
             speed = 0
         wait(10)
         print(speed)
         motor.run(speed)

dlech, 2020-06-23:
I am not able to reproduce the problem with the following script:

```python
from pybricks.pupdevices import Motor, ColorDistanceSensor
from pybricks.parameters import Port, Stop
from pybricks.tools import wait

motor = Motor(Port.D)
dial = Motor(Port.C)

while True:
      # Set the speed based on dial angle
     speed = dial.angle() * 3

     if abs(speed) < 100:
         speed = 0
     print(speed)
     motor.run(speed)
```

kueden, 2020-11-17:
Yes, this is exactly the same issue. The motors keep on running, but there is no way to stop it with the stop button.

dlech, 2021-04-02:
There have been significant changed made to Bluetooth in firmware v3.0.0b3. Can you please upgrade the firmware and test again?

kueden, 2021-04-03:
David
I got it to work, but there is another issue. There is a bug with the v3.0.0b3. I can connect via blootooth. But the play bottun (F5 for download and run) does not light up. I played around and got it to work. My original issue is resolved. However I cannot reproducibly get software to run since the play button does not show up. Eventually things run. I also learned playing with it that the "getchar" that I use in all my software (issue #154) is not available in this new version 

So it looks like my minor issue is resolved, but I cannot run my software any more due to the not active play buttion and the missing getchar function. Now it is very probable that this is my lack of competence. I tried on another computer. Same thing.

Thanks for all your help

Markus

laurensvalk, 2021-04-03:
See #287. It looks like this is affecting more people since a recent update.

closed 2021-04-26 by dlech