pybricks/support #37: Connecting to Pybricks BLE service fails on Android

laurensvalk, 2020-06-08:
Now that #34 has been fixed, something similar occurs on Android.

The bluetooth button appears to reach the connected state and it stays gray. After a while, it gives up:

[image]

I got the same error when initiating firmware update, but that did eventually succeed.

dlech, 2020-06-08:
I'm guessing that it had cached the old bad values from the older firmware. Everything is working for me on Android with the v3.0.0a2 firmware.

laurensvalk, 2020-06-08:
I'm not sure - this was after refreshing the cache and updating the firmware.

I'll try again on another Android device.

dlech, 2020-06-08:
I meant the OS Bluetooth device cache, not the browser cache.

dlech, 2020-12-11:
Is this still a problem?

laurensvalk, 2020-12-12:
I think we can close because we are not actively targeting Android right now.

closed 2020-12-12 by laurensvalk