pybricks/support #38: [Feature] 88005 Light Support

bricklife, 2020-06-08:
I tried to use 88005 Light from Pybricks, but it seems that the device is not supported.
https://www.lego.com/en-us/product/light-88005

LEGO official applications, e.g. Powered Up app, support it. I think it's almost the same as DCMotor. The IO Type ID is `0x08` as follows:
https://lego.github.io/lego-ble-wireless-protocol-docs/index.html#io-typ

laurensvalk, 2020-06-08:
Thank you for the request. Yes, we still have to add support for the Light.

Several others have asked the same, so it sounds like we should prioritize this one :smile:

laurensvalk, 2020-06-08:
The API will look like this.

[image]

bricklife, 2020-06-08:
Thank you for your quick response! The API looks good. I'm looking forward to it 🥰

Patlepirate01, 2020-06-08:
I was creating similar request when I saw this one...Test is on-going now well with 2 motors and light sensor...

laurensvalk, 2020-06-08:
Thanks! In the meantime, maybe you'll like to use the builtin light of the ColorDistanceSensor:

```python

my_sensor = ColorDistanceSensor(Port.A)
my_sensor.light.on(Color.BLUE)

```

Patlepirate01, 2020-06-08:
Thanks. Included in my current testing. Will probably do a small video later today

laurensvalk, 2020-06-08:
Preliminary API added in https://github.com/pybricks/pybricks-api/commit/80840396aadcfe839946ccf52242fa490b214f81

[---> Doc preview <---](https://docs.pybricks.com/en/work/pupdevices.html#light)

We'll keep this issue open until it is also implemented & released.

Thanks to @ZPhilo for the LDraw Parts!

ZPhilo, 2020-06-08:
> Thanks to @ZPhilo for the LDraw Parts!
You're welcome ;)
Just made an updated shortcut with realistic cable length (coiled)
https://www.ldraw.org/cgi-bin/ptdetail.cgi?f=parts/22168.dat
(don't forget the subpart! https://www.ldraw.org/cgi-bin/ptdetail.cgi?f=parts/s/22168s01.dat)

laurensvalk, 2020-06-08:
[image]

Patlepirate01, 2020-06-08:
Brilliant 🤓😇🥰

ZPhilo, 2020-06-09:
fiat lux...

laurensvalk, 2020-06-09:
Done! :bulb:  :rocket: :bulb: 

If you refresh Pybricks Code and update the firmware, light support is now included.

Run the example below for some fun tests.

**Also please note that from now on you have to include imports explicitly**

```python
from pybricks.pupdevices import Light
from pybricks.parameters import Port
from pybricks.tools import wait, StopWatch
from math import cos, pi

# Initialize the light and a StopWatch
light = Light(Port.B)
watch = StopWatch()

# Cosine pattern properties
PERIOD = 1
MAX = 100

# Make the brightness fade in and out
while True:
    # Get the time in seconds
    time = watch.time()/1000
    
    # Evaluate the cosine
    brightness = (0.5-0.5*cos(time*2*pi/PERIOD))*MAX
    
    # Turn on the light, wait, then repeat
    light.on(brightness)
    wait(10)
```

I'll close this issue when some of you have confirmed that it is working. Thanks!

bricklife, 2020-06-09:
@laurensvalk The example works fine! Thank you very much.

[image]

laurensvalk, 2020-06-09:
Excellent, thanks for confirming!

I'll close this issue.

closed 2020-06-09 by laurensvalk

Patlepirate01, 2020-06-09:
Works also well for me. Great team support. Really pleasant to test with a so great support