pybricks/support #39: SPIKE Ultrasonic sensor not returning expected results on CPlusHub [Bug]

Tcm0, 2020-06-08:
I tried to read the values of the SPIKE Prime ultrasonic sensor with the Control+ Hub but the results are inconsistend and/or don't make sense.
I understand that there is a bug in reading the sensor values in a loop (https://github.com/pybricks/support/issues/36) and this might be related to that, but it's another issue.

Code:
```
hub = CPlusHub()
ultraschallsensor = UltrasonicSensor(Port.A)

print(ultraschallsensor.distance())
while True:
    wait(100)
```

I ran the program mutliple times and moved my hand in front of the hub and the prints in the console don't seem to be consistent. It goes as far as reading higher values if the hand is closer to the sensor.

laurensvalk, 2020-06-08:
Have you tried pointing the sensor at a solid surface like a wall or a floor? The ultrasonic sensor is not really suitable for detecting small objects.

The values are the same as with the scratch app, only we don’t round to the nearest centimeter - we keep the original millimeter measurement. If you can verify that it is very different that the scratch app (except for a factor 10), then it might be a bug. Can you test this?

I take it you have placed your print statement in a loop? Reading sensors in a loop should be fine. The other issue is about printing thousands of messages per second. Just 10 a second should be fine.

Tcm0, 2020-06-08:
I think that the problem were wrong readings. Often I'll get the value 3000 which translates to 300cm and I think that that is the maximum value. It's probably something like "couldn't read".

laurensvalk, 2020-06-08:
Right, when it cannot detect anything, it returns the maximum possible distance.

That way, if you write a program that looks for objects, you don’t get false positives.

What is the maximum possible distance reported by Scratch?

Tcm0, 2020-06-08:
According to the official documentation sensor values can go from 5mm to 2000mm and 200cm is the maximum value in the spike prime app.

laurensvalk, 2020-06-08:
The sensor certainly reports values well above 200 cm, so that’s why our ceiling is currently at 300 cm.

Tcm0, 2020-06-08:
Maybe the sensor gets inaccurate for higher values, I don't know. The official documentation I linked lists 200cm/ 2000mm as the maximum distance.

laurensvalk, 2020-06-08:
That’s the marketing spec 😄

You can always cap the value to a lower threshold if you like.

Tcm0, 2020-06-08:
It wouldn't make sense to cap a value for marketing and I think that the EV3 sensor worked for 255cm. But they "guarantee" an accuracy of +/- 15 mm for the values up to 200cm in the documentation.
Anyways, that's just speculation :D

laurensvalk, 2020-07-10:
We reduced the "nothing detected" reading to 2000 to remove all doubt and updated the documentation accordingly. Thanks!

closed 2020-07-10 by laurensvalk