pybricks/support #40: [Bug] EV3 Brick system crashes while running my scan program

johnscary-ev3, 2020-06-09:
**Describe the bug**
I have a program that loops checking two color sensors, a sonic sensor, and gets IR button inputs.
While doing this it also moves a scan head motor using a generator (yield) function like in the Gyro demo program.
This program runs fine for a long time like an hour and then suddenly stops and causes the EV3 system to crash. Only can get it back with a hard reset.
I am using wifi connection but only an occasional print statement. No multi-threading in this app only that generator function to yield back control between motor moves spaced 1 sec apart.
I suspect some sort of memory leak but get no errors from the system.

Has this been seen before? Any ideas on how to diagnose this?

Thanks.

dlech, 2020-06-09:
You could open an ssh terminal and run `top` while your program is running to monitor memory usage. However, I don't think a memory leak would cause a complete system lock up.

Which mode are you using on the sensors?

You could also try running `sudo journalctl -b 1` to see if any errors were logged before the lock up on the previous boot.

johnscary-ev3, 2020-06-09:
Thanks for the inputs. Seems like running the "sudo" command wants a password which I don't know.
However I ran the "top" command fine.
I ran it before starting my program and then watched until I started to see some motor jerkiness and then system lock up. By that I mean the ssh connection stops and the brick is also not responsive any more from front panel and my program hangs.
My program loops through the sensors and does motor moves as I mentioned. The color sensors are doing color sensing and the sonic sensor is checking distance and I play a sound file if less than 300mm. 
Below is some data I recorded which seem to show memory being slowly consumed. It can take a hour or more for this to happen. Along the way I would occasionally (5 min) trigger a sound file from distance readings until the crash.

                    Before         @start       @motor_jerks      @crash
Mem used:      22K              24K          49K                    50K
Buff/cache:      34K              32K          6.3K                    6.1K
avail Mem:       32K              30K          4.5K                    3.7K

Any ideas on what I might be doing to use up the memory?

dlech, 2020-06-09:
> Seems like running the "sudo" command wants a password which I don't know.

`maker`

> Any ideas on what I might be doing to use up the memory?

Can you share the full program?

johnscary-ev3, 2020-06-11:
Hi,
The link below is the reduced code set that also shows the memory consumption.
I ran it both using the scan head motor and without using the motor. 
In the case without the motor, just sensors, it ran slighly longer but still got available memory down to 2.5K and then hung the system up.
The available memory will jump up and down, but trends down over the long haul.
Let me know what you think, or if you need something more.
Thanks for looking at this.

https://github.com/johnscary-ev3/ROBO_SCAN

johnscary-ev3, 2020-06-13:
Update:
I modified the ROBO_SCAN program to allow optical and sonic sensors, and motors, to be turned On/Off while running. I uploaded the new version with some comments on IR buttons to use.
After doing some testing using new program, it appears to me that the Optical Sensors doing Color readings have the most influence on memory loss.

I am fairly new to Python but I understand now that it generates some memory "trash" and then does "garbage collection" to clean it up. Seems like if I am generating trash faster than it can get cleaned up, I could see this memory loss affect. When I turn off the Optical sensors, I can see the lost memory come back.

Any inputs would be helpful. Thanks.

dlech, 2020-06-13:
EV3 sensors don't do very well when changing modes frequently. If possible, write your program using only one mode per sensor.

I also see you are potentially creating a new sensor class when there is an error. This should not be necessary. I suspect that this is leaking file handles. That could be possibly be causing the memory problems and system lock up. You can run `lsof | wc -l` to see how many files are open (have to install `lsof` package first).

johnscary-ev3, 2020-06-13:
Hi,
Thanks for inputs.
I will simplify my program by removing the error handling and any mode changing and see what happens.

I am trying to install lsof package. I am new to linux (debian) but seems like I should do this "sudo apt-get install lsof" but it does not find the package. I think the Linux system has access to ftp.debian.org  since I can ping it but does not seem to find package. Any guidance for me?

robot@ev3dev:~$ sudo apt-get install lsof
[sudo] password for robot: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package lsof is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

robot@ev3dev:~$ ping ftp.debian.org
PING ftp.debian.org(klecker-ftp.debian.org (2001:67c:2564:a119::148:12)) 56 data bytes
64 bytes from klecker-ftp.debian.org (2001:67c:2564:a119::148:12): icmp_seq=1 ttl=41 time=161 ms
64 bytes from klecker-ftp.debian.org (2001:67c:2564:a119::148:12): icmp_seq=2 ttl=41 time=161 ms
64 bytes from klecker-ftp.debian.org (2001:67c:2564:a119::148:12): icmp_seq=3 ttl=41 time=159 ms
64 bytes from klecker-ftp.debian.org (2001:67c:2564:a119::148:12): icmp_seq=4 ttl=41 time=161 ms

dlech, 2020-06-13:
Don't forget to run `sudo apt update` first/

johnscary-ev3, 2020-06-14:
Thanks for the tip on using "update". That did the trick and now I can run lsof command.
I have a number of years of experience running an engineering lab filled with Unix workstations and servers so most of Linux looks fairly familiar, except for the package installer apt ;0)

Anyway I made a number of changes to my  program to get rid of all the error handling, using just global variables, and checking the optical sensor calls. To make a long story short, it seems like the biggest thing was eliminating one of the two optical sensor calls I was doing. Not sure if doing those two calls back to back was a "mode switch" as you mentioned. But by doing just the "color" call and not the "rgb" call, the memory consumption seemed to get much better with avail mem staying at ~ 23K.
#color_rgb_int= cs.rgb()       
color_detected = cs.color()

I also ran the lsof | wc -l command and before starting my program it was 477, and while program running it is very stable at 882. So that does not seem to be a problem now I guess.
Anything else I should check?

Thanks.

BTW, I just saw that LEGO has announced a new version of Mindstorms for fall 2020.
On LEGO website it says that Python will be supported but not many details.
Is there more info on this floating around?

dlech, 2020-06-14:
On EV3, reading different values on sensors nearly always triggers a mode switch. 

If you run the "bad" program again and watch memory usage in `top` is it the `pybricks-micropython` process that grows or something else?

(sort by memory: https://unix.stackexchange.com/a/128957/91467)

johnscary-ev3, 2020-06-15:
Hi,
Link below is to some top data collected with "Good" program and "Bad" program.
Only difference is Good one does one optics sensor call and Bad one does both calls as mentioned above. Bad one eventually crashed.
Seems to me like main difference is with systemd-udevd process. Not sure what that is doing.
Any comments?

https://github.com/johnscary-ev3/Data_logs/blob/master/robo_scan%20mem%20tests.txt

dlech, 2020-06-15:
It is not looking ike a problem with `pbricks-micropython` at this point. Can you check the output of `dmesg` while the "bad" program is running.

I'm starting to suspect this could be an issue with the Linux kernel drivers for the sensors. e.g. https://github.com/ev3dev/ev3dev/issues/671

johnscary-ev3, 2020-06-16:
I took the dmesg data while running "bad" program but it looked the same as before running program.
But then I started to look more at those journal data and found some strange looking entries about the LEGO sensors at the end of a couple of those. Not sure if this is useful.
Let me know if there is something else I can collect to help.
Thanks

https://github.com/johnscary-ev3/Data_logs/blob/master/robo_scan_dmesg_data.txt
https://github.com/johnscary-ev3/Data_logs/blob/master/robo_scan_journal_data.txt
https://github.com/johnscary-ev3/Data_logs/blob/master/robo_scan_journal_data_1.txt

dlech, 2020-06-16:
The UART sensor issues are a known problem in ev3dev (see link in my previous comment). It needs to be fixed at the Linux kernel driver level so the best we can do for now is just try to avoid triggering it by not switching modes or using modes that have higher data rates (like RGB mode).

johnscary-ev3, 2020-06-16:
I did an additional experiment which was to use the color_rgb_int= cs.rgb() call instead of the color_detected = cs.color() call, reverse of my previous "good" test.
This worked ok also and gave very similar result of avail mem stable at ~23K.
So it seems for me the main issue was doing both calls back to back in my sensor loop.

Thanks so much for the support on this, and the coaching on Linux diagnostic  tools.
Keep up the great work on Pybricks!

dlech, 2020-06-17:
Thanks :grin:

laurensvalk, 2020-06-17:
Now that you've pinpointed the rootcause as the repeated mode switching, and now that you know `rgb()` works fine when used on its own, here's a suggestion so you can still have discrete color values.

It is fairly straightforward to covert the RGB values into HSV values. From there, it is fairly easy to get discrete color values by looking at the color hue. That way, you can still get both information types from just the RGB mode. We are doing something like this for some of the SPIKE sensors (and for the same reason too, we want to avoid mode switching).

And I just wanted to say thanks for giving us such detailed feedback. This is really helpful!

johnscary-ev3, 2020-06-17:
Hi,
That is a great idea to use the RGB values to get a color name!
I already found some Python code to do RGB to HSV but not having any luck finding py code to get the discrete colors from HSV. I found some tables for getting colors from HSV, but do you have a reference for some py code for getting basic "lego colors" from HSV data?
Thanks much.

laurensvalk, 2020-06-17:
If your application needs only "bright"/"saturated" colors, you can probably just use the H value.

You can then scan your objects beforehand and write down the H values. Then in your application you can measure H and "round" it to the nearest known value. If it's any help, here's where we do that in [C for the SPIKE sensor](https://github.com/pybricks/pybricks-micropython/blob/master/extmod/pbhsv.c#L45-#L86).

We could have added that to the EV3 too, but unfortunately the RGB values are not precalibrated, so everyone would get different results.

johnscary-ev3, 2020-06-18:
Hi,
Thanks for the reference code on getting colors from HSV data.
I implemented a shorter python version of your "error calculation" method for getting the colors.
It sort of works but I think , as you mentioned, the uncalibrated nature of RGB values makes it hard to get really good results particularly between sensors.
Since the normal color() call works fairly well, I assume it must take advantage of some internal cal factors?  I noticed that the B signal seems to have some slope and offset compared to the RG ones.
Is it possible to get cal factors from the sensor, if they exist? Thinking of doing my own cals using a black and white sample to get a slope and offset for linear correction of each channel, on each sensor.
What do you think?
BTW, my job before retirement involved measuring color of LEDs for characterization and production, including calibration of the equipment ;0)
Thanks.

laurensvalk, 2020-06-18:
> Since the normal color() call works fairly well, I assume it must take advantage of some internal cal factors? 

Yeah, we think so.

> Is it possible to get cal factors from the sensor, if they exist? 

Unfortunately they appear to be write-only :cry: See also https://github.com/ev3dev/ev3dev/issues/1386

You may be able to get better results using the `pybricks.iodevices.LUMPDevice` class and reading from `RGB-RAW`. Still uncalibrated though.

Indeed, you can try calibrate them yourself. I tried doing something like you described with the sensor at the time, but apparently the sensors vary a lot between produced units.

Curious to hear how it goes :)

johnscary-ev3, 2020-06-23:
Hi,
I noticed that raw rgb values are 300 to 350 for white paper @.25in so I picked 350 as default normalization and measured both sensors I have and got normalized rgb values.
Then I stuck these factors back into my normalization routine as shown below.
This allows both sensors to give similar values from same samples.
Doing this normalization I am able to get both sensors to measure LEGO RGBY brick samples and also am able to pick out Brown (my desk top) (red/yellow  h mid v) and Gray brick (mid v and not Brown).
Code below.
Thanks much

```

#Optical Sensor Normalization/Cal Factors 
# based on White Paper readings ~.25in. away to get all rgb values ~ 1.0
norm_default =350
#Sensor #1
norm_r_1 =norm_default*.88
norm_g_1 =norm_default*.95
norm_b_1 =norm_default*.99
#Sensor #2
norm_r_2 =norm_default*.82
norm_g_2 =norm_default*.76
norm_b_2 =norm_default*.69

def raw_to_rgb(sensor_number, r_raw, g_raw, b_raw):
    if sensor_number == 1:
        r, g, b = r_raw/norm_r_1, g_raw/norm_g_1, b_raw/norm_b_1
    elif sensor_number == 2:
        r, g, b = r_raw/norm_r_2, g_raw/norm_g_2, b_raw/norm_b_2
    else:
        r, g, b = r_raw/norm, g_raw/norm, b_raw/norm
    return r, g, b

class ColorEx:
    GRAY= "ColorEx.GRAY"

value_min     = 2
sat_threshold = 60
value_white   = 60
value_black   = 10

#Measured H midpoints using LEGO brick colors
hue_red = 350
hue_yellow = 30
hue_green = 160
hue_blue = 230
hue_brown_tol = 30

#Color.BLACK, Color.BLUE, Color.GREEN, Color.YELLOW, Color.RED, Color.WHITE, Color.BROWN
def hsv_to_color(h,s,v):
    color =None
    #If above min Saturation find closest match to colors
    if v < value_min:
        color = Color.BLACK
        if ColorDetectPrint:
            print("color=", color," V min hit=",'{:.2f} '.format(v))

    elif s > sat_threshold:
        #Calculate hue errors two ways for each color
        # (primarily needed for Red which can be on ether side of 0 point on color wheel)
        error =360
        error_test= abs(h - hue_red)
        if error_test < error:
            color = Color.RED
            error = error_test
        
        error_test= abs(h - (hue_red -360))
        if error_test < error:
            color = Color.RED
            error = error_test
        
        error_test= abs(h - hue_yellow)   
        if error_test < error:
            color = Color.YELLOW
            error = error_test

        error_test= abs(h - (hue_yellow -360))   
        if error_test < error:
            color = Color.YELLOW
            error = error_test

        error_test= abs(h - hue_green)  
        if error_test < error:
            color = Color.GREEN
            error = error_test

        error_test= abs(h - (hue_green -360))  
        if error_test < error:
            color = Color.GREEN
            error = error_test

        error_test= abs(h - hue_blue)  
        if error_test < error:
            color = Color.BLUE
            error = error_test
        
        error_test= abs(h - (hue_blue -360))  
        if error_test < error:
            color = Color.BLUE
            error = error_test

        if ColorDetectPrint:
            print("color=", color," H color check error=",'{:.2f} '.format(error))
    else:
        error = 100
        error_test=abs(v - value_white)
        if  error_test < error:
            color = Color.WHITE
            error = error_test
        
        error_test=abs(v - value_black)
        if error_test < error:
            color = Color.BLACK
            error = error_test

        
        #Special check for BROWN:  low s, mid-range v, and h near Red or Yellow
        error_test=abs(v- (value_white+value_black)/2 )
        if error_test < error :
            error = error_test
            if ( (abs(h - hue_red) <= hue_brown_tol) or (abs(h - hue_yellow) <= hue_brown_tol) ):
                color = Color.BROWN
            else:
                color = ColorEx.GRAY
        
        if ColorDetectPrint:
            print("color=", color," V color check error=",'{:.2f} '.format(error))
    
    return color

```

laurensvalk, 2020-10-09:
@dlech wrote:
> The UART sensor issues are a known problem in ev3dev (see link in my previous comment). It needs to be fixed at the Linux kernel driver level so the best we can do for now is just try to avoid triggering it by not switching modes or using modes that have higher data rates (like RGB mode).

@johnscary-ev3 wrote:
> So it seems for me the main issue was doing both calls back to back in my sensor loop.

With this answered and the issue known upstream (ev3dev), I think we can close this issue.

@johnscary-ev3 wrote:
> Thanks for the reference code on getting colors from HSV data.

We've been revising color detection pretty extensively on the other hubs. It could in principle be added to EV3 (https://github.com/pybricks/support/issues/138) but this is not currently a priority.

closed 2020-10-09 by laurensvalk