pybricks/support #41: [Question] [Control+] Random Number

GianCann, 2020-06-12:
The **random** module seems to be not present

```
import random
...
...
...

Traceback (most recent call last):
  File "main.py", line 5, in <module>
ImportError: can't import name random
```

dlech, 2020-06-12:
The `urandom` module is not currently enabled, but that is probably a useful one to enable if we have enough free space.

dlech, 2020-06-17:
`urandom` is enabled in https://github.com/pybricks/pybricks-micropython/commit/38b57c7e7683503140c8adfaaf95e006b70251f6 (all hubs except MoveHub).

Tip: no need to call `seed()`, this is done automatically when you import the `urandom` module.

closed 2020-06-17 by dlech

Patlepirate01, 2020-06-17:
Sorry for my question  but the status is not fully clear for me... is it now available or not ? And if yes how to do it ?

dlech, 2020-06-17:
We haven't updated the firmware at code.pybricks.com yet, but for those who can't wait, the latest builds are [here](https://github.com/pybricks/pybricks-micropython/actions?query=is%3Asuccess+workflow%3ABuild).

Click on a build (top is most recent) and then click on artifact to download it. Then open your downloads folder in a file browser. Click and drag the zip file to the firmware download button at code.pybricks.com.

Patlepirate01, 2020-06-17:
Thanks. I Will try. I would like to update my haunted house with the lights flashing as thunder and store it on the brick. I was already able to upload my own code on the brick...

Patlepirate01, 2020-06-17:
Ok. I was able to upload the last version and i'm now import uranium... Thanks again for your great and fast support