pybricks/support #42: [Question] Chrome 85: Cut & Paste

GianCann, 2020-06-12:
How can I copy (& paste in another place) the text from the output/log window?

CTLR+C don't work.

dlech, 2020-06-12:
Try <kbd>CTRL</kbd>+<kbd>INSERT</kbd> for copy and <kbd>CTRL</kbd>+<kbd>SHIFT</kbd>+<kbd>V</kbd> for paste (start the REPL then press <kbd>CTRL</kbd>+<kbd>E</kbd> for paste mode)

GianCann, 2020-06-12:
> Try CTRL+INSERT for copy and CTRL+SHIFT+V for paste

No, with this shortcuts don't work.
I want to" cut & past" error messages or script print output

dlech, 2020-06-12:
<kbd>CTRL</kbd>+<kbd>INSERT</kbd> works for me to copy from the terminal.

I just copied this:

```
('pybricks', 'cplushub', '3.0.0a3', 'v3.0.0a2-7-gb4f6e128 on 2020-06-10', 'cplushub')
```

GianCann, 2020-06-12:
Ok, yes... now it works.
I made an error with the laptop multifunctional keyboard.
Thank you @dlech

closed 2020-06-13 by GianCann

dlech, 2020-06-13:
FYI, there is now also a context menu when right-clicking in the terminal:
[image]