pybricks/support #43: Very often PyBricks reports that the sensor or motor is not connected

AlbanT, 2020-06-14:
Very often I get the message:

    Traceback (most recent call last):
      File "main.py", line 6, in <module>
    OSError: [Errno 19] ENODEV: 

    A sensor or motor is not connected to the specified port:
    --> Check the cables to each motor and sensor.
    --> Check the port settings in your script.
    --> Check the line in your script that matches
        the line number given in the 'Traceback' above.

I'm 100% sure all 3 motors I have are connected to the correct ports but still Pybricks complains that they are not connected correctly.
I have verified with Powered Up app that all 3 motors are functioning as desired eliminating the posibility for faulty connections.

The Code I'm using is very simple and straightforward:

    from pybricks.pupdevices import Motor
    from pybricks.parameters import Port, Stop
    from pybricks.tools import wait
    
    # Define the XYZ axis motor ports
    Xaxis = Motor(Port.C)
    Yaxis = Motor(Port.D)
    Zaxis = Motor(Port.B)
    
    # Say hello :)
    print("Lego 8094 says hello!")
    
    # First, we'll move the dial to zero.
    Xaxis.run_until_stalled(90,Stop.COAST,10)

I had this same issue last week and I got it working again by reflashing the official Lego firmware and reflash PyBricks again. Not sure if this actually fixed something or that it was coincidence that it started working again.

laurensvalk, 2020-06-14:
Hi @AlbanT ,

At the moment, you have to make sure sensors/motors are plugged in before you turn the hub on.

So, to change sensors/motors, just turn the hub off (long button press) and on again, and wait a few seconds for them to start up.

We are working to implement automatic detection of devices being plugged in or unplugged. This is just not done yet. There's no need to flash the firmware again. Putting the same software on it will not magically implement that somehow :)

I'll add an extra note to the documentation that this is still a work in progress, so others don't have to worry about it. Thanks for reporting!

TechnicBRICKs, 2020-06-20:
I stumbled on the same problem in my very first try of Pybricks! :)

> I'll add an extra note to the documentation that this is still a work in progress, so others don't have to worry about it. Thanks for reporting!

Where specifically I can find such notes?
Is there any beta release notes where we can see about current limitations overall (like supported/unsupported motors so far). I believe there is much more I should know, to avoid similar situations.
In my case in the sequence of this attempts I experienced severe browser freeze, or inability to connect to the hub, or at least run programs. Switching BT on/off was not enough, so I had to reboot my laptop every time I fall in similar situation.

laurensvalk, 2020-07-10:
> Where specifically I can find such notes?

We just added a note about sensors/motors on the [getting started page](https://docs.pybricks.com/en/latest/start_pup.html).

laurensvalk, 2020-12-11:
I think we finally fixed this! Does anyone want to help testing it before we merge it into the next release? If so, please try one or more of these firmwares:

- [Technic Hub](https://github.com/pybricks/pybricks-micropython/suites/1649065607/artifacts/30595557)
- [City Hub](https://github.com/pybricks/pybricks-micropython/suites/1649065607/artifacts/30595552)
- [Move Hub](https://github.com/pybricks/pybricks-micropython/suites/1649065607/artifacts/30595553)

To try them, see the `Saving a program permanently (BETA)` section in the [getting started guide](https://docs.pybricks.com/en/latest/start_pup.html).

BertLindeman, 2020-12-13:
Maybe this entry should be in a separate issue, I do not know.

Installed this firmware onto the movehub (as it is the most restricted hub)

At (or after?) the start of firmware flash an error on Chrome (Version 87.0.4280.88 (Official Build) (64-bit))  console:
```
index.js:511 Uncaught TypeError: S.current.contains is not a function
    at HTMLDocument.ee (index.js:511)
ee @ index.js:511
```

The web page looks like this:
[image]

The hub keeps blinking red-green-blue as if downloading the firmware and I awaited 20 minutes to no avail.

This is **not** the first time I installed this same firmware.
After an attempt to start the hub with the color sensor connected, it did not succeed.

The hub switched to steady blue but the web page stayed like this:
[image]

So had to report it...

Will re-try flashing the beta firmware.

BertLindeman, 2020-12-13:
[continuation] Back to regular firmware acts the same now. In both cases NO progress in progress-bar. No error at the moment like on flash of beta firmware.
Will re-load batteries (just did..) to be sure and re-try.

laurensvalk, 2020-12-13:
> Maybe this entry should be in a separate issue, I do not know.

Hub connectivity and firmware update issues are best reserved for separate issues indeed. Thanks for asking!

This issue is more about sensors/motors not working.

BertLindeman, 2020-12-13:
Will add new issue and link here.

BertLindeman, 2020-12-13:
Positive response from me.

using the movehub and playing with the Wedo motion sensor and the ColorDistance sensor on the C and D-port
while the hub is connected works fine.

dlech, 2021-01-07:
The firmware has changed quite a bit since this issue was opened and we're pretty sure it is fixed. Let's start a new issue if there are still related problems.

closed 2021-01-07 by dlech