pybricks/support #44: Flashing firmware to Technic hub gets stuck during flashing

AlbanT, 2020-06-14:
This is the 3th time in the past few hours I ran into this issue :(
This morning I tried flashing Pybricks to my Technic hub and halfway the progressbar would stop (the led still flashed Red Green Blue).
I removed the battery and retried only to have the same endresult :(

I flashed the official lego firmware back onto the hub and confirmed that it worked befor I retried to flash PyBricks again. This time it sucseeded. Unfortunately it would complain my motors weren't connected. This is other issue might and not be related but because of it I tried to flash Pybricks again and again the flashing stalls and seems to never finish...

Due to the experiences from the past I'm now flashing the original firmware and will try to flash pybricks after it finishes.

In the meantime the lego firmware flashed correctly and pybricks is flased as well.
Not sure what is heppening but it seems that flashing isn't as reliable as it should!?

laurensvalk, 2020-06-14:
Which operating system and which Chrome version are you using?

AlbanT, 2020-06-14:
> Which operating system and which Chrome version are you using?

- Windows 10 Pro version 1909 build 18363.836 

- Chrome version 85.0.4168.3 (Officiële build) dev (64-bits)

laurensvalk, 2020-06-14:
Thanks! We'll look into it. Please keep in mind that this is just at the alpha stage right now, and we're working hard to improve it.

Can you check with fresh batteries? Low battery can sometimes have this effect. In that case it may still work with the LEGO app because your phone is usually closer to the hub than your PC Bluetooth dongle.

Fortunately, the update procedure is securely stored on the hub no matter what, so there's no harm in retrying a few times.

laurensvalk, 2020-06-14:
As pointed out in the other issue, there's no reason to reflash anything. Here I'll just add that it also makes no difference to put the LEGO firmware back first. The existing firmware is not used during the update procedure, so that won't matter.

AlbanT, 2020-06-14:
> Can you check with fresh batteries? Low battery can sometimes have this effect. In that case it may still work with the LEGO app because your phone is usually closer to the hub than your PC Bluetooth dongle.

Unfortunately I have no fresh batteries around but the Lego app reports them as full and PyBricks returns around 6802mV
I'll keep this in mind when I run into it again. Also I will place the hub on top of my laptop to make sure the BT reception is very good :)

dlech, 2020-06-14:
Can you please check the console log and report back the last few errors listed there after it gets stuck?

If you need step-by-step instructions on how to do this: https://support.google.com/docs/thread/1873663?hl=en

AlbanT, 2020-06-21:
Sorry for the delay, been busy all week :(
Few days ago I was able to test it but Github was not available (Error 500).
Anyway when I was able to reproduce it (Today just now it worked fine) last Thursday the console log reported:
[image]

laurensvalk, 2020-06-28:
@martynboogaarts also had this issue. Same error in the log as the post directly above this.

This happened consistently. Also Windows 10.

I wonder if we might just need to slow down the process a bit.

dlech, 2020-07-03:
I have made some changes that could affect this. Could everyone please test again?

Walt-White, 2020-09-26:
Very similar situation flashing firmware to City hub.

64 bit Windows 10 Home Version 1909 OS Build 18363.1082
Chrome Version 85.0.4183.121 (Official Build) (64-bit)
ASUS laptop with internal Bluetooth radio.

Fresh batteries 9.51 volts.

Removed old device ("Smart hub") from list of known Windows Bluetooth devices that were ever paired.
Got through the firmware process up to seeing the red/green/blue sequence which lasted 6 minutes!

Hub started double blue flashing like it was looking for a device to pair with, as opposed to the instructions which say that the red/green/blue sequence ends with a solid blue light.

Web page now shows "Connect using Bluetooth" enabled.  Found Pybricks hub, clicked Pair, the hub is now solid blue.

Download/Run test program results in error about not seeing a motor on port A.

Looks like an incomplete firmware update process.

No new Chrome Console information.  There are three items which I think are remnants from the previous four hours of testing every permutation and combination of using a USB Bluetooth dongle (which temporarily disabled all Bluetooth on the laptop!), pairing with Windows before Pybricks, etc.:

DevTools failed to load SourceMap: Could not load content for chrome-extension://lgblkllcjgihfnlefhnnpppndbbjallh/scripts/vendor/browser-polyfill.js.map: HTTP error: status code 404, net::ERR_UNKNOWN_URL_SCHEME
DevTools failed to load SourceMap: Could not load content for chrome-extension://lgblkllcjgihfnlefhnnpppndbbjallh/scripts/vendor/hyperapp.js.map: HTTP error: status code 404, net::ERR_UNKNOWN_URL_SCHEME
DevTools failed to load SourceMap: Could not load content for chrome-extension://hiponeioelghhaljfflaaflpccedbdem/content-scripts/content-scripts/contentscript.js.map: HTTP error: status code 404, net::ERR_UNKNOWN_URL_SCHEME

laurensvalk, 2020-09-26:
> Hub started double blue flashing like it was looking for a device to pair with, as opposed to the instructions which say that the red/green/blue sequence ends with a solid blue light.
>
> Web page now shows "Connect using Bluetooth" enabled. Found Pybricks hub, clicked Pair, the hub is now solid blue.

This is the new expected behavior. We still have to update the getting started guide.

So the update was successful :)

> Download/Run test program results in error about not seeing a motor on port A.

Please make sure that the motors are plugged in before you turn the hub on. This is a known issue at the moment.

Walt-White, 2020-09-26:
Thanks for the quick response.  No change when ensuring that the motors are plugged in before turning on the hub.

But...…. just using the ColorDistanceSensor, I was able to turn the sensor's light to RED.  So, yes, it appears that the firmware was updated successfully.  Yay!

Now I'll have to try any other way to isolate why recognizing that there is a Motor connected to the hub is the problem.  I am importing Port in my program, just like the example.

Walt-White, 2020-09-26:
Update.  There's hope!  Using the example for [train motors](https://docs.pybricks.com/en/latest/pupdevices.html#motors-without-rotation-sensors), it works!

And to be honest, my project involves trains.  So I'm going to proceed with my project and let the experts ponder the Motor mystery.

Walt-White, 2020-09-26:
Update:  I should have been more specific.  The motor I'm connecting is the Simple Medium Linear Motor (goofy name, eh?) with LEGO Design ID 45303 and Bricklink part number 21980.  It's the motor that came with the City Hub in the PUP Batmobile.

Walt-White, 2020-09-27:
Well, it's another "pilot error."  I was using a Pybricks command for a motor that has a rotation sensor on a motor that doesn't have a rotation sensor.  I figured it out as I was crafting a thorough response to you, checking the documentation as I went.

Now using a DCMotor is the same, successful, for both the 45303 motor and the 88011 train motor.  Sorry to bother you guys.  I recommend closing this issue, at least for me.

CloudCCS, 2020-11-03:
I haven't seen an update here that solves this problem so I'd like to add my own experience. I have the Technic hub that came with the Volvo A60H. This hub(new batteries) gets stuck on the firmware update. This happens on every attempt. It has never worked. My boost hub does work. I'm new to pybricks, but I am a programmer and I have some time to give so if there is anything I can do to help, please ask. 

Windows 10 Pro 10.0.19041
Chrome 86.0.4240.183
Console:
io-6de156f3.js:111 Error: Failed to init: ,,true
    at flash-firmware.ts:302
    at l (runtime.js:63)
    at Generator._invoke (runtime.js:293)
    at Generator.next (runtime.js:118)
    at p (redux-saga-core.esm.js:1157)
    at c (redux-saga-core.esm.js:1251)
    at n (io-6de156f3.js:165)
    at Object.c [as cont] (redux-saga-core.esm.js:1251)
    at b (redux-saga-core.esm.js:1012)
    at Object.t.cont (redux-saga-core.esm.js:836)
b @ io-6de156f3.js:111
io-6de156f3.js:112 The above error occurred in task yr
    created by takeEvery(flashFirmware.action.flashFirmware, yr)
    created by wa
    created by wa
Tasks cancelled due to error:
takeEvery(flashFirmware.action.flashFirmware, yr)

dlech, 2020-11-17:
Hi @CloudCCS. If you still have some time and want to try to build Pybricks Code locally, you could try following the instructions [here](https://github.com/pybricks/pybricks-code/blob/master/CONTRIBUTING.md).

The code for flashing the firmware is in [this file](https://github.com/pybricks/pybricks-code/blob/master/src/sagas/flash-firmware.ts). You could try adjusting [this](https://github.com/pybricks/pybricks-code/blob/b9175543fd03c9c4e03869492acbca99748282a9/src/sagas/flash-firmware.ts#L322-L326) to see if you can get better results.

dlech, 2020-12-21:
I think this may have been the same issue as #178, which has been fixed. We can reopen or start a new issue if this is still a problem.

closed 2020-12-21 by dlech