pybricks/support #45: run_until_stalled doesn't allow duty_limit to be below16%

AlbanT, 2020-06-14:
This might be by design but if it is it isn't clear in the [documentation](https://docs.pybricks.com/en/latest/pupdevices.html#pybricks.pupdevices.Motor.run_until_stalled).

    Xaxis = Motor(Port.A)    
    Xaxis.run_until_stalled(360,Stop.COAST,15)

This results in:

    The requested operation is not valid in the current state:
    --> Check the documentation for required conditions.
    --> Check the line in your script that matches the line number given in the 'Traceback' above.

while `Xaxis.run_until_stalled(360,Stop.COAST,16)` works as expected.

laurensvalk, 2020-06-14:
It can't be lower than the feedforward signal (see pid settings). So to go lower, set the feed forward setting lower first, like 0.

This would be good to add to the docs indeed. Thanks!

laurensvalk, 2021-07-13:
The feed-forward setting has been removed a long time ago, so this issue no longer applies. Closing.

closed 2021-07-13 by laurensvalk