pybricks/support #46: Documentation about the imports and examples

AlbanT, 2020-06-14:
First of all I know this is alpha software so the documentation is not upto standard yet :)

I noticed that the program I used last week didn't include any `from pybricks.....` import lines.
As a reaction to Laurens his earlier reply about the duty limits for stalling. I tried to use the pid() command but it seems I need to import something for it but the current state of the documentation doesn't make it very clear for me when I need to import what :)

As this is more like support than beta testing I'll put the few support issues I run into into a single question, hope you agree this is best in order to avoid cluttering the issues :)

1. Looking at the [documentation of pid()](https://docs.pybricks.com/en/latest/motors.html?highlight=pid#pybricks._common.Control.pid) I think the usage is as follows for only retrieving the currently set values:

        Zaxis = Motor(Port.C)        
        Zaxis.Control.pid()
        print(kp, ", ", ki, ", ", kd, ", ", integral_range, ", ", integral_rate, ", ", feed_forward)

    but this results in : `AttributeError: 'Motor' object has no attribute 'Control'`

2. How can I drive 2 motors at the same time? I wonder if there is a function to drive 2 motors simultaniously making sure they both arrive at the set location (angle in my case) at the same time?
    Or at least the ability to issue a `run_target()` without waiting for it to finish.

Now I'm going to enjoy my sunday :p Thanks in advance!!! And keep up the good work! Hope I can show off my lego 8094 plotter driven by PyBricks soon :)

laurensvalk, 2020-06-14:
> I noticed that the program I used last week didn't include any from pybricks..... import lines.

Yep - you noticed correctly. They were automatic before, to reduce the program size a bit, but we decided it would make more sense to make it consistent with real Python programs. Especially when users want to import other things than the default.

> Looking at the documentation of pid() 

It's accessible through the [`control`](https://docs.pybricks.com/en/latest/pupdevices.html#pybricks.pupdevices.Motor.control) property. But I see now that its explanation is missing on the PUP page! For some reason it only shows up for EV3, but it is the same for both.

> How can I drive 2 motors at the same time? I wonder if there is a function to drive 2 motors simultaniously making sure they both arrive at the set location (angle in my case) at the same time?

It sounds like maybe you can use the [DriveBase](https://docs.pybricks.com/en/latest/robotics.html).

> Or at least the ability to issue a run_target() without waiting for it to finish.

You can set the argument `wait=False` to accomplish this.

> the documentation doesn't make it very clear for me when I need to import what :)

Yeah, we need to work on that. In general, the doc menu matches the modules exactly. You can also already check out the [EV3 examples](https://github.com/pybricks/pybricks-projects) for inspiration. The API/imports are the same.

> Hope I can show off my lego 8094 plotter driven by PyBricks soon :)

That would be awesome! The more example projects, the easier it will be for people to get started.

laurensvalk, 2020-06-14:
> It's accessible through the control property. But I see now that its explanation is missing on the PUP page! For some reason it only shows up for EV3, but it is the same for both.

This has been fixed now, so it should show up online when the build completes a few minutes. Thanks for reporting!

GianCann, 2020-06-14:
> Hope I can show off my lego 8094 plotter driven by PyBricks soon :)

My 8094-Pybricked need more improvement ;)
https://youtu.be/LrBrTQjghp0

laurensvalk, 2020-06-14:
Nice!

laurensvalk, 2020-10-07:
The examples now all have imports, so we can close this one. Thanks for reporting!

closed 2020-10-07 by laurensvalk