pybricks/support #47: [Bug] Couldn't flash "City Hub" for Pybricks

tomalphin, 2020-06-15:
**Describe the bug**
I am unable to update my City Hub to Pybricks.  This happened when the firmware on my City hub was old (last used around December 2019), and again after updating the firmware in the Powered Up app.

**To Reproduce**
Steps to reproduce the behavior:
1. at code.pybricks.com, click on the "City Hub" icon on the right nav bar.
2. scroll down, press "Click to show/hide the written steps" to see written instructions (I don't watch videos).
3. press-and-hold power on City Hub, and while still holding the button press icon with "flash hub firmware" tooltip in top navbar.
4. Select the "LEGO Bootloader" option in new window.

**Expected behavior**
5a. Flashing to begin

**Actual Behavior**
5b. Red error box (with rounded corners) indicating that the hub is not compatible. 

Error Text:  _"Sorry, we don't have firmware for this hub yet."_

**System Info**
Windows 10, 1909, Intel i7-9700K, 32gb RAM.
Chrome Dev: Version 85.0.4168.3 (Official Build) dev (64-bit)
[image]

**Screenshots**
[image]

GianCann, 2020-06-15:
City Hub is not supported at this moment.

laurensvalk, 2020-06-15:
City Hub firmware is partly ready. We mainly still need to:
- Activate/implement Bluetooth: This will be similar to Control+, so hopefully almost ready.
- Fix the sensor port UARTs

But these are key features to make it useful for most of users, so we've deliberately excluded it from the beta release.

jensadne, 2020-06-19:
Even though the City Hub firmware doesn't really work yet it should still be possible to flash the hub with it using the command line tool, shouldn't it? 

Flashing doesn't work for me, and after increasing the log level in ble.py to DEBUG I got the following, it looks like the connection is just dropped immediately? The LED on the hub does switch to flashing red-green-blue for a few seconds before going back to just flashing white.

```
Connected successfully!
Erasing flash and starting update
2020-06-19 18:30:15,593:    INFO: Getting device info.
2020-06-19 18:30:15,594:   DEBUG: TX CHUNK: bytearray(b'U')
2020-06-19 18:30:15,647:   DEBUG: Awaiting reply of 14
2020-06-19 18:30:17,900:    INFO: Disconnected by server.
```

And some hopefully relevant output from the bluetoothd log:
```
Jun 19 21:07:03 khentkawes bluetoothd[167409]: src/device.c:device_connect_le() Connection attempt to: 90:84:2B:2D:B3:82
Jun 19 21:07:03 khentkawes bluetoothd[167409]: src/adapter.c:connected_callback() hci0 device 90:84:2B:2D:B3:82 connected eir_len 31
Jun 19 21:07:03 khentkawes bluetoothd[167409]: attrib/gattrib.c:g_attrib_ref() 0x55f3d6364b40: g_attrib_ref=1
Jun 19 21:07:03 khentkawes bluetoothd[167409]: profiles/gap/gas.c:gap_accept() GAP profile accept (90:84:2B:2D:B3:82)
Jun 19 21:07:03 khentkawes bluetoothd[167409]: profiles/gap/gas.c:handle_characteristic() Unsupported characteristic: 00002a04-0000-1000-8000-00805f9b34fb
Jun 19 21:07:03 khentkawes bluetoothd[167409]: src/service.c:change_state() 0x55f3d63612d0: device 90:84:2B:2D:B3:82 profile gap-profile state changed: disconnected -> connected (0)
Jun 19 21:07:03 khentkawes bluetoothd[167409]: src/gatt-client.c:btd_gatt_client_connected() Device connected.
Jun 19 21:07:03 khentkawes bluetoothd[167409]: src/device.c:gatt_debug() MTU exchange complete, with MTU: 158
Jun 19 21:07:03 khentkawes bluetoothd[167409]: profiles/gap/gas.c:read_device_name_cb() GAP Device Name: LEGO Bootloader
Jun 19 21:07:03 khentkawes bluetoothd[167409]: profiles/gap/gas.c:read_appearance_cb() GAP Appearance: 0x1800
Jun 19 21:07:04 khentkawes bluetoothd[167409]: src/device.c:gatt_debug() Primary services found: 3
Jun 19 21:07:04 khentkawes bluetoothd[167409]: src/device.c:gatt_debug() start: 0x0001, end: 0x0004, uuid: 00001801-0000-1000-8000-00805f9b34fb
Jun 19 21:07:04 khentkawes bluetoothd[167409]: src/device.c:gatt_debug() start: 0x0005, end: 0x000b, uuid: 00001800-0000-1000-8000-00805f9b34fb
Jun 19 21:07:04 khentkawes bluetoothd[167409]: src/device.c:gatt_debug() start: 0x000c, end: 0x000f, uuid: 00001625-1212-efde-1623-785feabcd123
Jun 19 21:07:04 khentkawes bluetoothd[167409]: src/device.c:gatt_client_ready_cb() status: success, error: 10
Jun 19 21:07:04 khentkawes bluetoothd[167409]: src/gatt-client.c:btd_gatt_client_ready() GATT client ready
Jun 19 21:07:04 khentkawes bluetoothd[167409]: src/gatt-client.c:create_services() Exporting objects for GATT services: 90:84:2B:2D:B3:82
Jun 19 21:07:04 khentkawes bluetoothd[167409]: src/gatt-client.c:btd_gatt_client_ready() Features 0x00
Jun 19 21:07:04 khentkawes bluetoothd[167409]: src/gatt-client.c:btd_gatt_client_ready() Update Features 0x00
Jun 19 21:07:04 khentkawes bluetoothd[167409]: src/device.c:device_svc_resolved() /org/bluez/hci0/dev_90_84_2B_2D_B3_82 err 0
Jun 19 21:07:04 khentkawes bluetoothd[167409]: src/gatt-client.c:notify_client_ref() owner :1.708
Jun 19 21:07:04 khentkawes bluetoothd[167409]: src/device.c:gatt_debug() Registered handler for "Service Changed": 1
Jun 19 21:07:04 khentkawes bluetoothd[167409]: src/device.c:att_disconnected_cb()
Jun 19 21:07:04 khentkawes bluetoothd[167409]: src/device.c:att_disconnected_cb() Success (0)
Jun 19 21:07:04 khentkawes bluetoothd[167409]: src/service.c:change_state() 0x55f3d63612d0: device 90:84:2B:2D:B3:82 profile gap-profile state changed: connected -> disconnecting (0)
Jun 19 21:07:04 khentkawes bluetoothd[167409]: src/service.c:change_state() 0x55f3d63612d0: device 90:84:2B:2D:B3:82 profile gap-profile state changed: disconnecting -> disconnected (0)
Jun 19 21:07:04 khentkawes bluetoothd[167409]: src/gatt-client.c:btd_gatt_client_disconnected() Device disconnected. Cleaning up.
Jun 19 21:07:04 khentkawes bluetoothd[167409]: src/device.c:att_disconnected_cb() Automatic connection disabled
```

Very much open for the possibility that my bluetooth config is the problem here, so any suggestions for troubleshooting this are welcome. Did a quick test with bricknil, and that worked, so at least communicating with the stock firmware from this particular computer works.

dlech, 2020-06-19:
The city hub bootloader has some quirks that can cause BlueZ to disconnect. I haven't tested it yet, but I have a potential fix in progress at https://github.com/pybricks/pybricksdev/commit/3917215c599270cebe37f6993019174d05fd25a7

jensadne, 2020-06-20:
I can confirm that using the `macos` branch I was able to flash my city hub just fine.

GianCann, 2020-06-20:
@jensadne can you explain the step that you haved used?
I want to try too...
Thank you

jensadne, 2020-06-20:
I cloned the `pybricks-micropython` and `pybricksdev` repositories.
Installed `arm-none-eabi-gcc-cs-c++`, `arm-none-eabi-newlib` and `arm-none-eabi-binutils-cs` system packages, not sure all of them were needed, but that's what I have in my bash history. After that I could build the cityhub firmware with `make cityhub`.

Then I checked out the `macos` branch of `pybricksdev` and followed the instructions there. One thing to note is that in the `macos` branch the program name has changed to `pbdev` so the command to use for flashing is now `poetry run pbdev flash path/to/firmware.zip`

GianCann, 2020-06-21:
@jensadne I don't think I still have the ability to attempt a full firmware build, but I would like to attempt to upload the firmware to a City Hub
Can you provide me the firmware.zip file?

I want to use my Raspberry Pi 3B+ to upload the firmware.

laurensvalk, 2020-06-21:
There's no need to build the firmware - you can download it already just like the Control+ firmware builds.

GianCann, 2020-06-21:
@laurensvalk here there is only cplus and movehub, not city. Where I can find it?
https://github.com/pybricks/pybricks-micropython/releases/tag/v3.0.0a3

laurensvalk, 2020-06-21:
Please follow the usual link in the Powered Up Quickstart - not sure why you'd want want that specific tag.

GianCann, 2020-06-21:
This is my experience, with Windows 10 pro 64 Bit
- Git installed
- Python 3.8.3 installe
- Pybricksdev repository cloned
- Poentry installed
- Run `poentry install`:

```
C:\Users\Gianluca\pybricksdev>poetry install
Creating virtualenv pybricksdev in C:\Users\Gianluca\pybricksdev\.venv
Installing dependencies from lock file

Package operations: 67 installs, 0 updates, 0 removals

  - Installing decorator (4.4.2)
  - Installing ipython-genutils (0.2.0)
  - Installing six (1.15.0)
  - Installing attrs (19.3.0)
  - Installing parso (0.7.0)
  - Installing pyparsing (2.4.7)
  - Installing pyrsistent (0.16.0)
  - Installing pywin32 (228)
  - Installing traitlets (4.3.3)
  - Installing wcwidth (0.2.4)
  - Installing backcall (0.2.0)
  - Installing colorama (0.4.3)
  - Installing jedi (0.17.0)
  - Installing jsonschema (3.2.0)
  - Installing jupyter-core (4.6.3)
  - Installing markupsafe (1.1.1)
  - Installing packaging (20.4)
  - Installing pickleshare (0.7.5)
  - Installing prompt-toolkit (3.0.5)
  - Installing pygments (2.6.1)
  - Installing python-dateutil (2.8.1)
  - Installing pyzmq (19.0.1)
  - Installing tornado (6.0.4)
  - Installing webencodings (0.5.1)
  - Installing bleach (3.1.5)
  - Installing defusedxml (0.6.0)
  - Installing entrypoints (0.3)
  - Installing ipython (7.15.0)
  - Installing jinja2 (2.11.2)
  - Installing jupyter-client (6.1.3)
  - Installing mistune (0.8.4)
  - Installing nbformat (5.0.7)
  - Installing pandocfilters (1.4.2)
  - Installing pywinpty (0.5.7)
  - Installing testpath (0.4.4)
  - Installing ipykernel (5.3.0)
  - Installing nbconvert (5.6.1)
  - Installing prometheus-client (0.8.0)
  - Installing send2trash (1.5.0)
  - Installing terminado (0.8.3)
  - Installing notebook (6.0.3)
  - Installing pycparser (2.20)
  - Installing cffi (1.14.0)
  - Installing qtpy (1.9.0)
  - Installing widgetsnbextension (3.5.1)
  - Installing atomicwrites (1.4.0)
  - Installing cryptography (2.9.2)
  - Installing ipywidgets (7.5.1)
  - Installing jupyter-console (6.1.0)
  - Installing mccabe (0.6.1)
  - Installing more-itertools (8.4.0)
  - Installing pluggy (0.13.1)
  - Installing py (1.8.2)
  - Installing pycodestyle (2.6.0)
  - Installing pyflakes (2.2.0)
  - Installing pythonnet (2.5.0)
  - Installing qtconsole (4.7.4)
  - Installing asyncssh (2.2.1)
  - Installing bleak (0.6.4)
  - Installing flake8 (3.8.3)
  - Installing jupyter (1.0.0)
  - Installing mpy-cross (1.12)
  - Installing nbstripout (0.3.8)
  - Installing pybricks (3.0.0a3 3554814)
  - Installing pytest (5.4.3)
  - Installing tqdm (4.46.1)
  - Installing validators (0.15.0)
  - Installing pybricksdev (0.1.0)

C:\Users\Gianluca\pybricksdev>
```
After this, I have downloaded last build of cityhub-firmware.zip and cplushub-firmware.zip and copied this files on pybricksdev folder (without build with `make`, because I don't have the entire tool-chain to compile it by myself).

I put the CPlus Hub in Bootloader mode and I haved run the flash command, but there is not successful connection with the hub:

```
C:\Users\Gianluca\pybricksdev>poetry run pybricksdev flash cplushub-firmware.zip
Creating firmware
Searching for LEGO Bootloader
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\Gianluca\pybricksdev\pybricksdev\scripts.py", line 131, in entry
    _flash(args.arguments)
  File "C:\Users\Gianluca\pybricksdev\pybricksdev\scripts.py", line 106, in _flash
    asyncio.run(_main())
  File "c:\users\gianluca\appdata\local\programs\python\python38-32\lib\asyncio\runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "c:\users\gianluca\appdata\local\programs\python\python38-32\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "C:\Users\Gianluca\pybricksdev\pybricksdev\scripts.py", line 98, in _main
    address = await find_device('LEGO Bootloader', 15)
  File "C:\Users\Gianluca\pybricksdev\pybricksdev\ble.py", line 65, in find_device
    raise TimeoutError(
TimeoutError: Could not find LEGO Bootloader in 15 seconds

C:\Users\Gianluca\pybricksdev>
```

What did I miss doing?

laurensvalk, 2020-06-21:
`pybricksdev` is in very early development and not tested on all platforms yet, so it might be better to wait a while. In any case, a question like this would be better to have its new issue.

On topic: and the city hub is not even at the alpha stage at this point. With Bluetooth not yet enabled and the sensor ports mostly not working yet, there's little point in using this already :wink:

laurensvalk, 2020-06-21:
Still, for what it's worth, the City Hub fixes for flashing are now merged with the master branch on `pybricksdev`.

GianCann, 2020-06-21:
In the meantime I found the problem...
The BleakScanner don't return the name of BLE discovery device.
It return always 'Unknow', so it's impossible that this code can works well:

```
    # Create scanner object and register callback to raise discovery flag
    scanner = BleakScanner()
    scanner.register_detection_callback(set_device_discovered)

    # Start the scanner
    await scanner.start()

    INTERVAL = 0.1

    # Sleep until a device of interest is discovered. We cheat by using the
    # cross-platform get_discovered_devices() ahead of time, instead of waiting
    # for the whole discover() process to complete. We call it every time
    # a new device is detected by the register_detection_callback.
    for i in range(round(timeout/INTERVAL)):
        # If device_discovered flag is raised, check if it's the right one.
        if device_discovered:
            print("Device discovered")
            # Unset the flag so we only check if raised again.
            device_discovered = False
            # Check if any of the devices found so far has the expected name.
            devices = await scanner.get_discovered_devices()
            for dev in devices:
                print(dev)
                print(dev.name)
                # If the name matches, stop scanning and return address.
                if name == dev.name:
                    await scanner.stop()
                    return dev.address
        # Await until we check again.
        await asyncio.sleep(INTERVAL)

```

laurensvalk, 2020-06-21:
As mentioned above, please ask unrelated questions in a new issue.

dlech, 2020-07-03:
City hub support is now in beta. Be sure to unplug everything from the port before flashing the firmware.

closed 2020-07-03 by dlech