pybricks/support #48: Pybricks v3 Beta Announcments

dlech, 2020-06-16:
Subscribe to this issue to be notified of new features and bug fixes during the initial v3.0 beta period.

Changes are published on https://code.pybricks.com as soon as they are merged.

See https://github.com/pybricks/support/issues/29 for the v3.0 roadmap.

dlech, 2020-07-03:
https://code.pybricks.com has been updated.

Added:
- Support for City hub (note, requires Chrome 85 to flash firmware due to quirks in the City hub bootloader)
- Changed firmware flashing packet size on City hub and Control+ hub (speeds up flash time)
- Increased frequency of progress checks when flashing firmware (slows down flash time but should help prevent errors)
- Firmware v3.0.0a4
  - Bluetooth support on City hub
  - Fixed sensor ID issue on City hub
  - WeDo 2.0 sensor support
  - Fixed Move hub motor polarity
  - Enabled builtin `help` and `input` functions
  - Enabled `urandom` module

dlech, 2020-07-04:
https://code.pybricks.com has been updated.

Fixed:
- Not working in offline mode.

dlech, 2020-07-15:
https://code.pybricks.com has been updated.

Fixed:
- `cityhub` code snippet indentation
Changed:
- Firmware v3.0.0a5
  - Added `ColorDistanceSensor.hsv` method
  - Sensor bug fixes

dlech, 2020-07-18:
https://code.pybricks.com has been updated.

Added *Clear* button to terminal context menu

[image]

dlech, 2020-08-05:
https://code.pybricks.com has been updated.

Changed:

- Firmware v3.0.0a6
  - Lots of refactoring (if there are any regressions, this is probably the cause)
  - Added experimental `Matrix` class for linear algebra (not available on Move hub)
  - Added colors `Color.CYAN`, `Color.MAGENTA`, `Color.GRAY`
  - Renamed `Color.PURPLE` to `Color.VIOLET`
  - Removed `uos.uname()`
  - Added `from pybricks import version`
  - Adjust Control+ gyro filter ([so you can do this](https://youtu.be/qi6MRyLSrsI))

dlech, 2020-09-04:
https://code.pybricks.com has been updated.

Changed:

- Firmware v3.0.0a8
  - New/changed APIs for hub status light
  - `pybricks.parameters.Color` now represents HSV colorspace values instead of just being the names of colors
  - Fix Control+ hub low battery shutdown at boot   
  - Update to MicroPython v1.13 (not a major jump since we already were based on v1.12 plus >600 commits)

dlech, 2020-10-10:
https://code.pybricks.com has been updated.

Changed:

- Firmware v3.0.0a9
  - **Breaking change: `CPlusHub` class is renamed to `TechnicHub`**
  - Breaking change: color sensor color map API changed
  - Fixed Technic hub and City hub not connecting on Windows

nuno2500, 2020-10-10:
Here are the windows updates. Hope it helps. Thanks .

[image]
[image]

Walt-White, 2020-10-13:
I have two computers with two very different behaviors.  The easiest to describe is my desktop:
Lenovo Razor ideacentre Y900 RE-341SZ Signature Edition
Windows 10 Home version 2004 build 19041.508
Chrome browser Version 85.0.4183.121 (Official Build) (64-bit)

When I go to code.pybricks.com and hit <CTRL> Refresh to be sure I have the latest firmware, and try to download to my City Hub, I get the message "No compatible devices found."

I've tried to pair the City Hub outside of Pybricks by just using the Windows "Add Bluetooth Device" utility, and the City Hub is never seen.  So it's like my internal Bluetooth radio isn't compatible with the City Hub.  When I look at Bluetooth in Device Manager, there are a lot of things listed and I can't tell which of them is the internal Bluetooth radio.

I'll list my laptop behavior in #136 because it's similar to things discussed there.

dlech, 2020-11-09:
https://code.pybricks.com has been updated.

Changed:

- Firmware v3.0.0a10
  - Fix connecting to City Hub on Windows not working
  - Add `pybricks.experimental.getch` function for reading on character from stdin
  - Move `Matrix` class from `pybricks.experimental` to `pybricks.geometry`
  - Various `Matrix` class fixes and improvements
  - Battery voltage and current measurement improvements

dlech, 2020-11-21:
https://code.pybricks.com has been updated.

Changed:

- Firmware v3.0.0a11
  - Added Technic Angular motor IDs
  - Fixed lights not turning off on shutdown until button is released (powering off I/O ports on shutdown is still TODO)
  - Changed motor PWM frequency
  - Proper license file included in binary distribution

dlech, 2020-12-11:
https://code.pybricks.com has been updated.

Changed:

- Fixed terminal losing focus after opening context menu
- Added a few more keyboard shortcuts

dlech, 2020-12-11:
https://code.pybricks.com has been updated.

Changed:

- Firmware v3.0.0a12
  - Reduce heap size on BOOST Move hub to prevent crash
  - Fix stop button not working in Pybricks Code when using REPL
  - Turn off Bluetooth when running program without already being connected to Bluetooth
  - Turn of hub if no Bluetooth connection was made or program not started within 30 seconds
  - Change program running light indication to breathing pattern instead of steady on

dlech, 2020-12-18:
https://code.pybricks.com has been updated.

Fixed:

- Sometimes hub does not reboot after flashing firmware (depending on firmware size)

dlech, 2021-01-08:
https://code.pybricks.com has been updated.

Fixed:

- Firmware v3.0.0a12 
  - Changed program running indication light
  - Enabled built-in reversed() function
  - Fixed hotplug of UART I/O devices
  - New PUPDevice.info() method
  - Increased auto-shutdown timeout to 3 minutes
  - Moved IMU from experimental to hub class
  - IMU improvements on Technic hub

dlech, 2021-01-15:
https://code.pybricks.com has been updated.

Added:
- New settings panel
- New "dark mode" setting
- New "flash my program" setting (known issue: browser has to be refreshed if you try to flash a program with compile errors)
- Tip on how to zoom in/out

Changed:
- Documentation toggle is moved to settings
- Buttons are rearranged a bit

dlech, 2021-01-17:
https://code.pybricks.com has been updated.

Added:
- Help section with links to support sites
- About dialog

dlech, 2021-01-22:
https://code.pybricks.com has been updated.

Added:
- Link to example projects

Changed:
- Update to new version without having to restart browser or clear cache (this change won't be visible until after the update - i.e. on the next update)
- Compiler error notification is removed when it is no longer applicable (i.e when code is compiled again)
- Progress indicator moved from status bar to firmware flash button
- Firmware v3.0.0b1:
  - IMU improvements on Technic hub
  - Fix status light bug on Technic hub

dlech, 2021-01-23:
https://code.pybricks.com has been updated.

Fixed:
- Improved error handling when flashing firmware.

dlech, 2021-01-24:
https://code.pybricks.com has been updated (v1.0.0-beta.3).

Added:
- Added *Select All* to context menus

Fixed:
- Allow selecting text in dialogs and notifications.

dlech, 2021-01-26:
https://code.pybricks.com has been updated (v1.0.0-beta.4).

Fixed:
- Removed extra space under icon in buttons.

Changed:
- Docs are now abridged version of https://docs.pybricks.com (EV3 removed).
- Docs now work when using the app offline.

dlech, 2021-01-29:
https://code.pybricks.com has been updated (v1.0.0-beta.5).

Fixed:
- Settings button should not be selected after closing settings.
- Documentation  does not remember position after hide/show.

Added:
- App install button.
- App check for update button.
- Ctrl-D keyboard shortcut to toggle docs.

dlech, 2021-02-19:
https://code.pybricks.com has been updated (v1.0.0-beta.7).

Changed:
- Firmware v3.0.0b2
  - Fix I/O device detection on Technic hub
  - Minor Bluetooth improvements
  - `Color` type API changes
  - Move hub IMU improvements
  - Update to Micropython 1.14

dlech, 2021-04-02:
https://code.pybricks.com has been updated (v1.0.0-beta.8).

This is likely the last beta before we issue a release candidate, so please test!

Changed:
- Firmware v3.0.0bb
  - Enabled `uio`, `usys` and `uselect` modules (except on Move hub)
  - Removed experimental `getchar()` function. Use `usys.stdin` with `uselect.poll` instead.
  - Enabled stack check to avoid hub lock-up on stack overflow.
  - Significant Bluetooth improvements.
  - Fix issues with red and green LEDs on Technic hub.
  - Automatic import of `pybricks.*` in REPL.
  - Add `Color.NONE` enum value (Existing uses of `None` should be replaced with this.)
  - Stop button (hub button and IDE button) now raise `SystemExit` instead of `KeyboardInterrupt`.
  - Pressing the hub button will now stop the REPL if it is running.
  - Added `pybricks.experimental.set_stop_button()` to change or disable the stop button so that it can be used by the user program.

dlech, 2021-04-03:
v1.0.0-beta.8 has serious Bluetooth issues on Windows, so we have reverted to v1.0.0-beta.7 for now.

dlech, 2021-04-05:
https://code.pybricks.com has been updated (v1.0.0-beta.9).

Changed:

- Firmware v3.0.0b4:
  - Added workaround for Bluetooth status notification issue on Windows (#287)

dlech, 2021-04-06:
https://code.pybricks.com has been updated.

Fixed:
- Download and run of programs not working (regression due to firmware change in previous release).

dlech, 2021-04-07:
https://code.pybricks.com has been updated (v1.0.0-beta.11).

Fixed:
- Download and run of programs sometimes not working (#290).

dlech, 2021-04-07:
https://code.pybricks.com has been updated (v1.0.0-beta.12).

Fixed:
- Fix regression from v1.0.0-beta.11 that caused the run button to become disabled after running a small program

dlech, 2021-04-12:
https://code.pybricks.com has been updated (v1.0.0-beta.13).

Fixed:
- Fix buttons getting stuck disabled when downloading and running a program
- Fix extra line breaks in notifications with action buttons
Changed:
- Disable firmware flash button when hub is connected
- Updated documentation
- Firmware v3.0.0b5
  - Removed pre/post program run messages

dlech, 2021-04-12:
https://code.pybricks.com has been updated (v1.0.0-beta.14).

Added:
- Progress indicator when downloading the program.

dlech, 2021-04-26:
https://code.pybricks.com has been updated (v1.0.0-beta.14).

Changed:
- Updated to latest docs
- Firmware update to v3.0.0b6:
  - Fixed Bluetooth lockup on Technic and City hubs
  - Fixed `Color` attributes being garbage collected

dlech, 2021-05-11:
https://code.pybricks.com has been updated (v1.0.0-rc.1).

Fixed:
- Fixed issue with extra scroll bars in the terminal pane.

Changed:
- Firmware v3.0.0c1:
  - Implemented watchdog timer on all hubs (should reboot instead of freeze so you don't have to pull the battery anymore).
  - Fixed status light not always turning off when powering off.
  - Fixed some lockups related to frequent UART writes
  - Fixed some lockups related to frequent Bluetooth writes.
  - Improved `DriveBase` default settings.

dlech, 2021-06-08:
Latest changelog can now be found at https://github.com/pybricks/pybricks-micropython/blob/master/CHANGELOG.md.

closed 2021-06-08 by dlech