pybricks/support #49: [Question] Multiple Hub codind?

GianCann, 2020-06-16:
I tried to use two separate instances of Chrome, trying to connect to two different Technic Hubs simultaneously (both already with Pybricks on board). With the first instance, I can connect to the first Hub, while the second cannot "see" the second Hub.

laurensvalk, 2020-06-17:
The online Pybricks Code app will probably only support one hub for now. This is aimed at beginning users. We'd like to make this work really well before we start thinking about next steps.

**But don't worry -- something is already on the way for more advanced users!**

It will let you run multiple hubs at once. And so you can even make them talk to your PC as well. So even though it will take a while before we enable hub-to-hub communication, hub-to-pc-to-hub is already possible. Give us a day or two to clean it up. If you want to, familiarize yourself with `jupyter-notebook` in the meantime. :wink:

laurensvalk, 2020-10-07:
This is now possible with [pybricksdev](https://github.com/pybricks/pybricksdev). Feel free to open a new issue if there are any questions.

closed 2020-10-07 by laurensvalk