pybricks/support #50: [Bug] Flash new firmware with custom main.py don't work

GianCann, 2020-06-16:
I tried to load the firmware with a custom main.py, following the steps explained on the documentation (drag the new .zip file on 'flash firmeare' button).
After drag the zip file, the flash firmware button don't work anymore, with no reaction to click.

GianCann, 2020-06-16:
I have used the v3.0.0a2 firmware 
https://github.com/pybricks/pybricks-micropython/releases/tag/v3.0.0a2

dlech, 2020-06-16:
Can you please check the console log and report back the last few errors listed there after it gets stuck?

If you need step-by-step instructions on how to do this: https://support.google.com/docs/thread/1873663?hl=en

GianCann, 2020-06-16:
I found the error thanks to the console log: the zip contained a sub folder and so the firmware-base.bin has not viewed from the code.pybricks.com
Now I'm flashing the firmware with my custom main.py

GianCann, 2020-06-16:
Ok, firmware loaded with my custom main.py but the program don't start when I press the green button.
This is the program (the light is connected before power on):
```
from pybricks.pupdevices import Light
from pybricks.parameters import Port
from pybricks.tools import wait, StopWatch
from math import cos, pi

# Initialize the light and a StopWatch
light = Light(Port.A)
watch = StopWatch()

# Cosine pattern properties
PERIOD = 2
MAX = 100

# Make the brightness fade in and out
while True:
    # Get the time in seconds
    time = watch.time()/1000
    
    # Evaluate the cosine
    brightness = (0.5-0.5*cos(time*2*pi/PERIOD))*MAX
    
    # Turn on the light, wait, then repeat
    light.on(brightness)
    wait(10)
```

dlech, 2020-06-16:
Does the program work when running from the web browser?

GianCann, 2020-06-16:
I think there is a firmware mismatch:

```
Traceback (most recent call last):
  File "main.py", line 1, in <module>
ImportError: no module named 'pybricks.pupdevices'
```

GianCann, 2020-06-16:
> Does the program work when running from the web browser?

I have two Technic Hub flashed with Pybrick:
- the first one flashed with the 'built-in' firmware come with code.pybricks.com: on this hub, the program (from the code editor, works fine)
- the secondo one flashed with "my" firmware composed by .bin and .json from https://github.com/pybricks/pybricks-micropython/releases/tag/v3.0.0a2, and main.py that contain the "Light" program: with this firmware, the code editor return the error **ImportError: no module named 'pybricks.pupdevices'**

dlech, 2020-06-16:
It looks like we missed tagging `v3.0.0a3`. I just started a build, so check the GitHub releases again in a few minutes.

GianCann, 2020-06-16:
Yeah! It's work! ;)
Thank you @dlech

closed 2020-06-17 by GianCann