pybricks/support #51: [Question] Read serial input buffer

GianCann, 2020-06-16:
In the 'autonomous running' mode, there is a way to read the input buffer from the nRF UART Service?
I can "print" on this service, so I can send data to other device... but if i want to receive some byte from the 'external world'?

https://youtu.be/s1B9pz1sM9U

dlech, 2020-06-16:
It looks like we need to enable `MICROPY_PY_BUILTINS_INPUT` to get the builtin `input()` function.

laurensvalk, 2020-06-17:
This is available in the latest build using:

```python
input()
```

closed 2020-06-17 by laurensvalk

GianCann, 2020-06-17:
What am I doing wrong?

```
a=input()
print(a)
```

_('pybricks', 'cplushub', '3.0.0a3', 'b8aea4e on 2020-06-09', 'cplushub')
Traceback (most recent call last):
  File "main.py", line 7, in <module>
NameError: name 'input' isn't defined_

GianCann, 2020-06-17:
Ah... It's not yet released?

https://github.com/pybricks/pybricks-micropython/commit/38b57c7e7683503140c8adfaaf95e006b70251f6

laurensvalk, 2020-06-17:
You can always download the latest builds.

Click on the green check mark of any version you want, usually the latest [from master](https://github.com/pybricks/pybricks-micropython/commits/master):

[image]

Then click on details and then pick the artifact you want to download:

[image]

GianCann, 2020-06-17:
@laurensvalk It works, but not exactly as I expected.
I meant a non-blocking input/read function.
However, this is already useful as is

laurensvalk, 2020-06-17:
That will be there in time. We might have a `Mailbox`, similar to EV3 (see docs). Would that suit your use case?

GianCann, 2020-06-17:
> Would that suit your use case?

I try to explain one use case.
Think of a train circuit with 3 different trains.
Each train has its own pre-established schedule
Each train is viewed as a BLE serial device by another system (Arduino? Raspberry? ...)
By sending a specific command to every single train, I could change the default behavior of the main program or maybe activate a secondary one

laurensvalk, 2020-06-17:
That sounds like the perfect use case for the Mailbox approach :+1:

GianCann, 2020-06-17:
There is a strange behavior...

Look this very simple code:

```
from pybricks.pupdevices import Light
from pybricks.hubs import CPlusHub
from pybricks.parameters import Port
from pybricks.tools import wait, StopWatch
hub = CPlusHub()
import uos

# Initialize the light and a StopWatch
lightA = Light(Port.A)
lightB = Light(Port.B)

print(hub.battery.voltage())
print(uos.uname())

x=True
s=False 
while x==True:
    a=input()
    if a=="a1":
        lightA.on(100)
    elif a=="a0":
        lightA.on(0)
    elif a=="b1":
        lightB.on(100)
    elif a=="b0":
        lightB.on(0)
    elif a=="end":
        s=True
        x=False
    elif a=="close":
        x=False
    else:
        print(a + " not a valid command")

print("Close program")
if s==True:
    print("Shutdown the hub")
    wait(500)
    hub.shutdown()
```
1) If I run the code from the App, It works perfectly
2) If I merge the main.py with the firmware, It work perfectly using the App only as 'terminal' (running the main.py from the green button)
3) If I connect the HUB with another BLE UART client (as nRF UART App) It doesn't work... I get the initial print output (voltage battery and firmware version) but if I send a command ("a1", "end"...), the hub reply with same string, without react to the command (power on the Light or shutdown the hub).

[image]

laurensvalk, 2020-06-17:
We don't currently support this custom interface - please check in a few days where we are with `pybricksdev`. You can learn how things work from there.

GianCann, 2020-06-17:
ok, I thought serial communication was two-way anyway.

dlech, 2020-06-17:
The input function reads until there is a newline character before continuing. I don't remember if the nRF app lets you send a newline character or not.

GianCann, 2020-06-17:
> I don't remember if the nRF app lets you send a newline character or not.

I also tried with nRF Toolbox, which allows you to specify how to close the transmission: LF, CR or LF + CR

GianCann, 2020-06-17:
I obtained it works with a my Android App that send the string with a \n\r termination!