pybricks/support #52: [Feature] Hub name

GianCann, 2020-06-17:
Add a method/property to the hub class to set/get the name of the hub.
This name is useful to identify the hub when searched by BLE scan.

Nelvarion, 2021-01-08:
@dlech @laurensvalk 
Is this on the roadmap?
If not, do we have another method of identifying the hub? I'm trying to reuse a script with different behavior according to the hub it's running on.

laurensvalk, 2021-01-08:
Yes, see #173 for an option to choose the name when you install the firmware. Would that work?

In terms of self-identification, we can probably also make its own Bluetooth address readable.

Nelvarion, 2021-01-08:
+1000 for naming the hubs!
So right now there's no way to identify the hub from within a script?

laurensvalk, 2021-01-08:
Are they both the same type? If they're different, you could `try/except`  to import `MoveHub` or `CityHub`.

Two other options:
- Build from source and change the name
- Take the last and second to last firmware version, and read the firmware version (`pybricks.version`) to tell them apart :smile:

Nelvarion, 2021-01-08:
Working with `CityHubs`, so can't use the type.
I'd use the second option but I need the `experimental` class and it was added in the last version 😉 
I'll work around it for now until you add naming.

laurensvalk, 2021-01-08:
There have been a lot of [builds since then](https://github.com/pybricks/pybricks-micropython/actions?query=workflow%3ABuild+branch%3Amaster+is%3Asuccess).

dlech, 2021-06-26:
Setting hub name is now possible with https://github.com/pybricks/pybricksdev/releases/tag/v1.0.0-alpha.10 and firmware >= v3.1.0a1.

closed 2021-11-20 by dlech