pybricks/support #54: [Question] Show main.py

GianCann, 2020-06-17:
It's possible to view the content of a "stored" (main.py) program in a PU hub?

laurensvalk, 2020-06-17:
No, that's not possible as it is not stored on the hub in plain text.

The program gets compiled using mpy-cross before being included in the firmware. The same holds for programs you download-and run.

GianCann, 2020-06-17:
OK, thank you for the clarification.

laurensvalk, 2020-06-17:
So before I close this issue, how do you like the new permanent program feature? :smile:

Eventually, it will be integrated in the Pybricks Code app so it's much more convenient to do, but I hope you can live with this for now.

GianCann, 2020-06-17:
> So before I close this issue, how do you like the new permanent program feature?

For me IS THE KEY FEATURE of Pybrick ;)

> Eventually, it will be integrated in the Pybricks Code app so it's much more convenient to do, but I hope you can live with this for now.

Yes, that's fine in this mode.
The main.py stored in the firmware is an operation to do occasionally, and not at every run.
So, it's not matter to not have it in the App

laurensvalk, 2020-06-17:
Cool, thanks!

closed 2020-06-17 by laurensvalk

GianCann, 2020-06-18:
@laurensvalk it's possible to add a "title/name" in the json file and get this string in some way at runtime, to obtain a reference to the program stored with the firmware?

laurensvalk, 2020-06-18:
It's not *im*possible.

How about this at the top of your program instead? Will that be just as useful?

```python
print('I am the train program! Written by GianCann on June 18, 2020!')
print('Train motor should be on Port A!')
```

GianCann, 2020-06-18:
> Will that be just as useful?

Sometimes you have the solution under your nose, but you don't see it. :grimacing: 

Thank you @laurensvalk  ;)

laurensvalk, 2020-06-18:
You're welcome. Much easier for us this way too :smile: :rocket: