pybricks/support #55: Not able to upload firmware on Imac[Bug]

Patlepirate01, 2020-06-17:
**Describe the bug**
I try to reinstall last version of the firmware (after using again standard lego firmware in the past days) without success

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Flash HUB firmware'
2. I push the green button, select the hub in the list and associate the Hub
3. Light is flashin Green, Red, Blue,.... as expected but no upload....
4. I stop after 10'...

**Expected behavior**
Firmware upload

dlech, 2020-06-17:
Can you please check the console log and report back the last few errors listed there after it gets stuck?

If you need step-by-step instructions on how to do this: https://support.google.com/docs/thread/1873663?hl=en

dlech, 2020-06-17:
Possibly same problem as  #44.

Patlepirate01, 2020-06-17:
Will check later. In the meantime I have tried on my other iMac and it seems to be working...

Patlepirate01, 2020-06-17:
Would be good to have a message at the end of the firmware upload.... Updated successful (if it's the case I didn't receive it...)

dlech, 2020-06-17:
We are working on it :smile: Also working on better error notification.

Patlepirate01, 2020-06-17:
Not able to get the checksum...

[image]

Patlepirate01, 2020-06-17:
Working iMac: OS Catalina 10.15.5 with Chrome 85.4.41.73.0
Non working Mac Book Air with : OS High Sierra 10.13.6 with Chrome 85.4.41.73.0

dlech, 2020-07-03:
I have made some changes that could affect this. Could everyone please test again?

Patlepirate01, 2020-07-03:
It works. I have tested the upload on Cplus and City hubs and both are working well... It's a good present for my Holidays that will start in a few hours.... Will have more time for testing .

Patlepirate01, 2020-07-03:
For me issue is closed

closed 2020-07-03 by Patlepirate01