pybricks/support #56: [Bug]

GianCann, 2020-06-17:
Sometime, when try to connect to the hub, this error happened:
[image]

This is the report from console:
DOMException: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
kn @ error-log.ts:20

Making a second attempt, it works

dlech, 2020-10-10:
If this happens again, can you please collect Bluetooth logs and post them here?

- Open an administrator command prompt (search start menu for `cmd` then right click and select "Run As Administrator")
- Change to the directory where you want to save the file (e.g. `cd %HOMEPATH%/Downloads`)
- Run the command: 

      logman create trace "bth_hci" -ow -o bth_hci.etl -p {8a1f9517-3a8c-4a9e-a018-4f17a200f277} 0xffffffffffffffff 0xff -nb 16 16 -bs 1024 -mode Circular -f bincirc -max 4096 -ets
- Reproduce your issue.
- Run the command: 

      logman stop "bth_hci" -ets

- Open the directory in File Explorer
- Right-click the file and select "Send To > Compressed (zipped) Folder"
- Upload the zip file here

mwinkler, 2020-11-03:
Hi
I have the same issue with a LEGO City hub (Windows 10 - 2004 / Chrome 86).
Flashing works without a issue, but I cannot connect after.

Here the logman zip:
[bth_hci.zip](https://github.com/pybricks/support/files/5484097/bth_hci.zip)

dlech, 2020-11-03:
City hub is a known issue: https://github.com/pybricks/support/issues/48#issuecomment-706463541

dlech, 2021-01-29:
We've made some changes around this, so lets start a new issue if the same error message is seen again.

closed 2021-01-29 by dlech

xblanch, 2021-03-08:
Hi,
I ave the same problem with City Hub. I have hub flashed but now I can't connect.
```
GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
```
Thanks.

dlech, 2021-03-09:
Hi @xblanch, can you please start a new issue with more details on your OS, web browser and what you did that lead up to the problem?

vdespa, 2024-07-19:
In case it helps anyone: 

I also encountered this error. It turned out that the batteries were at about 1V each. After replacing them, it worked without any issues.