pybricks/support #59: [Bug]

ekmanb, 2020-06-19:
I can't get drivebase to work. 
robot = DriveBase(left_motor, right_motor, WHEEL_DIAMETER, AXLE_TRACK)
always give ValueError: Invalid argument

I tried lots of rebuilds and get stopped. Any ideas?

**Screenshots**
[image]

laurensvalk, 2020-06-19:
Which motors are you using? Are they both the same type?

Can you check on the EV3 menu which motors you see in the device browser?

laurensvalk, 2020-06-19:
You have to use either two Large Motors, or two Medium motors.

Sometimes they can detected incorrectly when you plug them in. Unplugging the cable and plugging it back helps in that case.

ekmanb, 2020-06-19:
I am using two Large Motors and the cables are well connected. I know what you described could be a problem, but not this time. Is there another EV3 MicroPython other than pybricks?

laurensvalk, 2020-06-20:
They can get detected incorrectly even if they are well connected. So have you tried unplugging and then running again?

And if it still doesn’t work, what does the device browser from the main menu report?

ekmanb, 2020-06-21:
Laurens,
Thanks for hanging in there. I tried resetting the cables at again, and this time it seemed to correct the issue. I am now going forward with many of the examples. I am mentoring several middle school students remotely, given we have canceled our summer robotics camps and are still in partial lock-down.

ekmanb, 2020-07-01:
We continue having issues with motor statements, especially using old NXT large motors. We reset the cables and the problem goes away for a few days. We seldom have this issue using the same EV3 with LEGO Mindstorms software. Is there something we are missing? We were planning to use Python in a robotics class, but the problem is so random and unpredictable, we are rethinking the class.

The Labview software reports out the motors and sensors on a small window when we connect to the brick. Are there any tools that show the port usage before we run our code?

laurensvalk, 2020-07-02:
As mentioned a few times above, could you check the device browser from the main menu? You can find instructions in the starters guide / online api. I think this is what you are asking for?

At least as a workaround for now, you can check if everything is there and unplug if not. You will only need to do this just after starting up. Once a sensor/motor is properly detected, it stays that way.

I’m sorry to hear that this problem is happening. We appreciate your feedback - it would be helpful to get the device information mentioned above so we have a bit more detail to work from.

laurensvalk, 2020-07-02:
@dlech, have you seen this in ev3dev lately?

dlech, 2020-07-02:
Yes, it happens sometimes.

Indiphile, 2021-11-24:
> You have to use either two Large Motors, or two Medium motors.
> 
> Sometimes they can detected incorrectly when you plug them in. Unplugging the cable and plugging it back helps in that case.

It's working.

GuyNicol, 2022-05-18:
I had a similar problem (1 large and 1 medium motor) and the various comments found here helped me resolve the issue. In my case, I'm using 3 motors and I had a mismatch in my cables and I was setting up the DriveBase with 1 large and 1 medium motors instead of the two large motors. Thanks!

dlech, 2025-07-13:
We have dropped support for ev3dev so we can close this issue.

closed 2025-07-13 by dlech