pybricks/support #60: [Feature] Configurable color sensor threshold

Nelvarion, 2020-06-19:
**Is your feature request related to a problem? Please describe.**
The [PoweredUp color & distance sensor](https://www.lego.com/en-us/product/color-distance-sensor-88007) gives out inconsistent reading for some of the supported colors.
Red, yellow and white are usually detected properly at a distance of a few studs.
However, the other colors are either very hard to match or are often detected as other colors by the sensor.
For example:
White detected as yellow
Blue and green are often cross detected.

From my experience with both the PoweredUp mobile app, and the [The Brick Automation Project](https://www.facebook.com/pages/category/Games-Toys/The-Brick-Automation-Project-2428195644074254/), the app gets back a color value (RGB? HEX? Color name?) and a "certainty" measurement, only values above a set "certainty" threshold are considered a correct match. 

This makes programming a layout that's meant to be displayed in public prone to reading errors, and coding all possible workaround complicates the code considerably.

**Describe the solution you'd like**
The ability to control the threshold at which a color is considered matched to allow for more flexibility with color detection - given environmental constraints such as ambient light that can affect detection.

**Describe alternatives you've considered**
Changing the direction of the sensor (facing sideways/downwards) helps remove some variables from the equation, namely the ambient light, but can still be hit or miss.
[Different sensor directions on a train](https://imgur.com/a/VJcfHZG)

**Additional context**
This request is an attempt to better control a train layout with colored signals (tiles/bricks) with which the code alters the train(s) behavior.
Slow at a yellow, stop at red, etc.
Working with only 3 colors makes the code much more complicated than if we had the full range of colors reliable - currently storing values in hub.state[] slots to account for any branching.

laurensvalk, 2020-06-19:
Thanks for your request. We talked earlier briefly, and I mentioned that we had some ideas for this. In fact, we have already done something like this for the SPIKE Color Sensor. The documentation for that SPIKE class [is here](https://docs.pybricks.com/en/latest/pupdevices.html#color-sensor) and I've posted an example below.

Does this meet your use case?

Then we can add it to the ColorDistanceSensor too.

Naturally, we cannot change how well the sensor works, but you can make it more reliable by choosing only the colors you need in your application. All the others then get rounded to the nearest match, so the range can improve drastically. For example, if you don't mind not having cyan or purple, then detection of red, blue and green should be much more reliable.

Is there something you would rather see differently? The class we made for the SPIKE Color Sensor isn't final yet, so we're happy to adapt it based on user feedback.

We like to use HSV instead of RGB since it is a bit more intuitive. You can easily find descriptions online. I think this [picture from wikipedia summarizes it nicely](https://commons.wikimedia.org/wiki/File:HSV_color_solid_cylinder.png).

#### Example (Short version):

```python
sensor = ColorSensor(Port.A)

# Configure the sensor to detect only red, blue, white, or none.
sensor.color_map({Color.RED: 350, Color.BLUE: 210}, 30, {None: 0, Color.WHITE: 60})
```

#### Example (As above, just with more explanation):

```python
from pybricks.pupdevices import ColorSensor
from pybricks.parameters import Port, Color
from pybricks.tools import wait

sensor = ColorSensor(Port.A)

# You can measure the HSV values like this. You can measure this for your
# application for use in the settings below.
print(sensor.hsv())

# Let's say we are only interested in red and blue, so we could do:
hues = {
    Color.RED: 350,
    Color.BLUE: 210
}

# We want the saturation to be at least 30% before we consider it a real color.
saturation = 30

# If the saturation is less than that, it is either black, white, or nothing.
# Which one it is depends on the birghtness value (V)
# Again, you can choose to include only what is relevant in your application.
values = {
    None: 0,
    Color.WHITE: 60,
}

# Now let's store these settings
sensor.color_map(hues, saturation, values)

# Now, in the rest of your program, the sensor will only detect blue and red
# if the saturation is high enough. Otherwise, it reports white or none.

while True:
    print(sensor.color())
    wait(1000)

```

Nelvarion, 2020-06-20:
This seems to cover all scenarios I can think of right now, will give it a go once City hubs are supported and reopen if needed.
Thanks for the quick reply and good luck with the project!

laurensvalk, 2020-06-20:
Sounds good!

No need to close yet - this has yet to be added to the ColorDistanceSensor.

laurensvalk, 2020-07-07:
The latest build (download the zip and flash it manually) has support for this in the ColorDistanceSensor. You can read `hsv` too for even more detail.

We would love to hear your feedback if you try it, so we can improve it based on your findings.

Here's a basic example to try it. You can modify desired colors using the techniques discussed above, but the default should already give a much improved detectable range for the primary colors.

```python
from pybricks.pupdevices import ColorDistanceSensor
from pybricks.parameters import Port
from pybricks.tools import wait

sensor = ColorDistanceSensor(Port.A)

while True:
    print(sensor.color())
    wait(100)
```

Also tagging @GianCann who might want to try this example.

Nelvarion, 2020-07-07:
Awesome, thank you!
Will try to give it a go tonight, already have a script ready.

laurensvalk, 2020-07-07:
[Just added the `hsv` and `color_map` to the docs](https://docs.pybricks.com/en/latest/pupdevices.html#color-and-distance-sensor).

laurensvalk, 2020-07-07:
To make sure you are running the latest version, you can simply check that this method is already available:

```python
sensor = ColorDistanceSensor(Port.A)
print(sensor.color_map())
```

GianCann, 2020-07-07:
> Also tagging @GianCann who might want to try this example.

👍 

({Color.RED: 350, Color.YELLOW: 30, Color.GREEN: 110, Color.BLUE: 210}, 50, {Color.BLACK: 10, Color.WHITE: 60, None: 0})

Traceback (most recent call last):
  File "main.py", line 9, in <module>
KeyboardInterrupt:

laurensvalk, 2020-07-07:
I meant the example [I was talking about when I wrote it ](https://github.com/pybricks/support/issues/60#issuecomment-654806808) :smile: .

GianCann, 2020-07-07:
Works well also with two sensor :smile: 
[image]

laurensvalk, 2020-07-07:
Cool. I found that if you put them too close together they tend to interfere.

GianCann, 2020-07-07:
The right distance for a good reading is about one stud, but it also depends on the ambient light.

laurensvalk, 2020-07-07:
I think you'll find that the right distance can be much bigger with this firmare update. Perhaps compare with the original app. (Until now, we gave you the same processed color value, but now we calculate our own to get better range).

GianCann, 2020-07-07:
> I think you'll find that the right distance can be much bigger with this firmare update

Yes! It's true... about 7/8 studs for the red brick! Over this distance, the sensor return Yellow or none

laurensvalk, 2020-07-07:
> Yes! It's true...

:tada: :rocket: :tada:

laurensvalk, 2020-07-07:
And you can stretch it even further if you want. For example, if you have only red and blue in your application, you can use the example above to ignore all other colors, so it would not even report yellow at the edge.

The defaults are probably fine for most applications though.

Nelvarion, 2020-07-07:
> > I think you'll find that the right distance can be much bigger with this firmare update
> 
> Yes! It's true... about 7/8 studs for the red brick! Over this distance, the sensor return Yellow or none

Need to come up with use cases for such distances, that's great.

laurensvalk, 2020-07-07:
It does get less accurate if you get *that* far out, haha.

The goal was to improve accuracy at short range, but some people will always stretch it to the limit :smile:  (Which we totally encourage by the way! Can't wait to see what everyone will come up with.)

laurensvalk, 2020-07-07:
Tagging @ZPhilo too, who asked about the time it takes to get a new color. I think the `color()` method might be a bit faster too now in this mode, but I haven't actually tried this.

JorgePe, 2020-07-07:
need to test this again reading Color+ hub light... some weeks ago was testing crazy idea of of using light for communication between hubs but never got consistent readings.

laurensvalk, 2020-07-07:
You weren't trying to read the color emitted by the other sensor by any chance?

JorgePe, 2020-07-07:
I hope not... I made a contraption to put the color-distance sensor over the hub light in the best position with a 2L cylinder to enclose light. And gave up, BLUE was read as WHITE, RED as YELLOW, YELLOW also as YELLOW none of them was 100% proof.
Then turned to read ambient() with the sensor almost touching the light... best readings were when light was not RED, ORANGE or BROWN.

JorgePe, 2020-07-07:
and then ordered a SPIKE color sensor :D

laurensvalk, 2020-07-07:
> I hope not... I made a contraption to put the color-distance sensor over the hub light in the best position with a 2L cylinder to enclose light. And gave up, BLUE was read as WHITE, RED as YELLOW, YELLOW also as YELLOW none of them was 100% proof.

The ColorDistanceSensor reads color by emitting some light and reading how much of that comes back. It has no way to measure colors like you were asking of it :smile: 

Hence the suggestion to use ambient. Not to sense color, but simply good old on/off keying.

By contrast, the SPIKE Color Sensor is more like a "1-pixel camera" with optional "flash light". So there you can read ambient color. This is enabled in the Pybricks API by choosing `surface=False`.

Supposedly that sensor has a much better range, but with our firmware the ColorDistanceSensor is pretty good too, especially if you consider its price and other functionalities.

JorgePe, 2020-07-07:
not just the price... it's the only available through LEGO site. SPIKE sensors or motors are only available (individually) through EDUCATION partner... and there are only 2 local partners in my country, with a long history of bad service... and LEGO doesn't allow to order from foreign partners. So if you are on a budget its Bricklink or just BOOST parts.

GianCann, 2020-07-07:
> it's the only available through LEGO site. SPIKE sensors or motors are only available (individually) through EDUCATION partner... and there are only 2 local partners in my country, with a long history of bad service... and LEGO doesn't allow to order from foreign partners

In Italy there is only ONE Educational partner, so...
[image]

Nelvarion, 2020-07-07:
I'm getting consistent RED readouts at 10 studs away!
Going to switch the sensor to face down (it's mounted on a train) to see how well it's behaving while driving.

Nelvarion, 2020-07-07:
[image]

Getting excellent reading on red, green, blue, yellow and white at a motor.dc rate of 60 on OK batteries - that's a good place for a display run.
The sensor loops every 50ms.
All of this at default colormap values.
On our last show I had to get really creative with variables since I could only work with red, yellow and white reliably - it's gotten so much easier now!

laurensvalk, 2020-07-08:
That's really fantastic to hear! I'm really happy that the effort has paid off with good results :)

By the way - in case you are curious, we are using the RAW RGB mode of the sensor, convert it to HSV, and then find the best `Color` match based on the user-specified (or default) `color_map()`. The source is available, but this is the gist of it.

We'll be releasing a few train videos in the next few days. They were made a long time ago... and I'm starting to wonder which mode we used. I seem to remember we used RGB as well for the same reason, but we had to hard code all the values because we hadn't made the nice `hsv()` and `color()` conversion that we have now :smile: .

I'll probably close this issue in a day or two, then, but feel free to keep posting your findings here. Closed issues will still be visible and Google will often send a few people this way.

I'll also prepare some guidelines/instructions for sharing scripts at [pybricks-projects](https://github.com/pybricks/pybricks-projects). It would be cool if you want to share a basic script for that train (e.g. timed), as well as one with sensors. For the sensors, it would be nice to have both a really simple one (for example, go back and forth between blue and red), and one that is as complex as you want.

Nelvarion, 2020-07-08:
My python skills are quite basic, but I can help with a few simple scripts.
Will gladly share.

laurensvalk, 2020-07-08:
Simple is good, really :+1: . We'd love to make Pybricks accessible for non-programmers too, or anyone completely new to Python.

Complex scripts can be fun, but they can be intimidating to new users, or anyone just learning a text-based language.

ZPhilo, 2020-07-09:
> 
> 
> Tagging @ZPhilo too, who asked about the time it takes to get a new color. I think the `color()` method might be a bit faster too now in this mode, but I haven't actually tried this.

Was not home... I'll try to have a look (but I believe that speed is intrinsic to sensor hardware!)

laurensvalk, 2020-07-09:
We use a different mode now, which does physically less (It does rgb instead of rgb+ir), so in theory it might be faster. Again, no promises, just a hunch :)

You can tell the difference because the lights appear much brighter than by default, because it spends no time doing invisible IR.

I also noticed that in some (other) sensors that they do some form of hold/hysteresis. If the SPEC1 mode did that too (I don’t know if it does, but it could), then we have gotten rid of that too, which would also contribute to faster readings.

ZPhilo, 2020-07-09:
> 
> The latest build (download the zip and flash it manually) 
...and where do I find it? sorry to be a burden, but Github remains like a deep dark forest to me...
Overwise, I tested again (with the old firmware obviously!) and I didn't find Boost sensor to be that slow - actually I found its speed performace similar to that of Spike Prime sensor. Need to do some more tests ;)

laurensvalk, 2020-07-09:
You can find the link in the same instructions about saving your script permanently, but here it goes:

https://github.com/pybricks/pybricks-micropython/commits/master

Pick the latest (so at the top) and click on the green checkmark, then click on `details`.

Then pick the file from the artifacts:

[image]

(Don't worry about the indicated size - it's actually bigger once you download it.)

laurensvalk, 2020-07-09:
And slow/fast depends a bit on what we are talking about: are we talking about how fast `color()` executes, or how soon it detects a new color in your ball sorter? My discussion about speed above was based on the latter, because I thought you said that was somewhat slow.

ZPhilo, 2020-07-10:
> You can find the link in the same instructions about saving your script permanently, but here it goes:

Ah yes, indeed. Least than can be said is that it's not self explanatory, and vocabulary used by Github doesn't help (Artifacts  ????).

> My discussion about speed above was based on the latter, because I thought you said that was somewhat slow.

I talked about overall speed (number of times a moving colored object could be detected correctly in a given time, using a tight loop).

laurensvalk, 2020-07-10:
Yeah, sorry about the complexity at this stage. Eventually, we'd like both the `main.py` and the firmware version to be simply selectable from our online Pybricks Code app. (For example, so you can choose between `stable` and `latest` firmware).

> I talked about overall speed

Right :+1: Looking forward to hear how that goes if you try it.

laurensvalk, 2020-07-10:
I just updated those instructions, thanks @ZPhilo !

laurensvalk, 2020-07-11:
This has now been implemented, documented, tested by quite a few of you, and we've just added examples as well. I think we can call this done and close the issue.

Thanks for the input everyone!

For most users there's this:

[image]

But for you there's this:
[image]

closed 2020-07-11 by laurensvalk