pybricks/support #62: [Question]: Delay between different Infrared Sensor modes

TheVinhLuong102, 2020-06-21:
Hi @laurensvalk @dlech I have programmed the Ev3rstorm with both PyBricks2 and EV3Dev2 and would like to report significantly slower execution by PyBricks2 than by EV3Dev2. My 2 programs are attached for your review.

[Ev3rstorm-PyBricks2-vs-EV3Dev2.zip](https://github.com/pybricks/pybricks-api/files/4809179/Ev3rstorm-PyBricks2-vs-EV3Dev2.zip)

The 2 programs are trying to perform the same capabilities:
- Driving around by the IR Beacon
- Detect objects by the IR Sensor
- Blast the bazooka when the Touch Sensor is pressed/bumped

Each of these activities is conditioned on certain "if"s, and the activities are arranged sequentially in a main "while True" loop. No multiprocessing/threading is used.

When I execute the EV3Dev2 program, the performance is pretty smooth: the Ev3rstorm is very responsive to sensory stimuli and actuates the various tricks satisfactorily.

However, when I run the PyBricks2 program, the Ev3rstorm's responses are very laggy/slow, suggesting that each loop in the "while True" takes a significantly longer time to execute.

Could you let me know how I can go about troubleshooting this? Thank you @laurensvalk @dlech .

TheVinhLuong102, 2020-06-21:
Thanks @laurensvalk. I was aware of https://github.com/ev3dev/ev3dev/issues/1401 but didn't make the mental connection :) I can confirm now that removing one of the IR-related activities made the delay go away in my PyBricks Ev3rstorm.

Are you considering adding your recommendations above into the PyBricks documentation? These issues and the corresponding technical decisions are not obvious to most users, and others may encounter difficulties from time to time if they are unaware.

laurensvalk, 2020-06-21:
Yes, so let's keep this issue open until we've added that, thanks.

laurensvalk, 2020-07-10:
Documented in https://github.com/pybricks/pybricks-api.

Thanks for reporting!

closed 2020-07-10 by laurensvalk