pybricks/support #63: [Question] Compile script error

GianCann, 2020-06-21:
I tried this example [explained here](https://github.com/pybricks/pybricksdev/tree/master#compiling-pybricks-micropython-programs-without-running) and I get **unrecognized arguments** error:

```
C:\Users\Gianluca\pybricksdev>poetry run pybricksdev compile 'print("Hello!"); print("world!");'
usage: pybricksdev compile [-h] script
pybricksdev compile: error: unrecognized arguments: print(world!);'
```

Without the `print("world!");` part, It works

```
C:\Users\Gianluca\pybricksdev>poetry run pybricksdev compile 'print("Hello!");'

Bytes:
b'M\x05\x00\x1f $\x00\n\x00\x07\x1abuild\\_tmp.py\x00Qc\x00\x00'

// MPY file. Version: 5. Size: 29
const uint8_t script[] =
    0x4D, 0x05, 0x00, 0x1F, 0x20, 0x24, 0x00, 0x0A,
    0x00, 0x07, 0x1A, 0x62, 0x75, 0x69, 0x6C, 0x64,
    0x5C, 0x5F, 0x74, 0x6D, 0x70, 0x2E, 0x70, 0x79,
    0x00, 0x51, 0x63, 0x00, 0x00,
};

C:\Users\Gianluca\pybricksdev>
```

Why?

laurensvalk, 2020-06-21:
I wonder if Windows deals with spaces differently here. Can you try it without the space in before the second print?

GianCann, 2020-06-21:
Yes, without space it works:
```
C:\Users\Gianluca\pybricksdev>poetry run pybricksdev compile 'print("Hello!");print("world!");'

Bytes:
b'M\x05\x00\x1f $\x00\n\x00\x07\x1abuild\\_tmp.py\x00Qc\x00\x00'

// MPY file. Version: 5. Size: 29
const uint8_t script[] =
    0x4D, 0x05, 0x00, 0x1F, 0x20, 0x24, 0x00, 0x0A,
    0x00, 0x07, 0x1A, 0x62, 0x75, 0x69, 0x6C, 0x64,
    0x5C, 0x5F, 0x74, 0x6D, 0x70, 0x2E, 0x70, 0x79,
    0x00, 0x51, 0x63, 0x00, 0x00,
};

C:\Users\Gianluca\pybricksdev>
```

dlech, 2020-06-21:
The space isn't the problem, the single quotes are. CMD on Windows does not treat single quotes as a special character. This should work better:

    poetry run pybricksdev compile "print('Hello!'); print('world!');"

GianCann, 2020-06-21:
Yes, right!

```
C:\Users\Gianluca\pybricksdev>poetry run pybricksdev compile "print('Hello!'); print('world!');"

Bytes:
b'M\x05\x00\x1f l\x08\n\x00\x07\x1abuild\\_tmp.py\x00\x11\x00{\x10\x0cHello!4\x01Y\x11\x00{\x10\x0cworld!4\x01YQc\x00\x00'

// MPY file. Version: 5. Size: 57
const uint8_t script[] =
    0x4D, 0x05, 0x00, 0x1F, 0x20, 0x6C, 0x08, 0x0A,
    0x00, 0x07, 0x1A, 0x62, 0x75, 0x69, 0x6C, 0x64,
    0x5C, 0x5F, 0x74, 0x6D, 0x70, 0x2E, 0x70, 0x79,
    0x00, 0x11, 0x00, 0x7B, 0x10, 0x0C, 0x48, 0x65,
    0x6C, 0x6C, 0x6F, 0x21, 0x34, 0x01, 0x59, 0x11,
    0x00, 0x7B, 0x10, 0x0C, 0x77, 0x6F, 0x72, 0x6C,
    0x64, 0x21, 0x34, 0x01, 0x59, 0x51, 0x63, 0x00,
    0x00,
};

C:\Users\Gianluca\pybricksdev>
```